3KUdelta/Solar_WiFi_Weather_Station #1: Issue reading 16 MB Flash on Wemos D1 Mini Pro

Lefuneste83, 2019-05-20:
Hi there. I have tried uploading your sketch on a Wemos D1 Mini Pro using Arudino IDE 1.8.9.
SPIFFS seems to have a problem accessing the 16 MB flash memory, which seems to be a common issue according to my research on forums.
By specifing a Wemos D1 R2 & Mini in the Arduino IDE, I manage to have everything working just fine.
I have looked around for some hints but the issue seems quite general with 16 MB flash chips.
How did you manage to get yours working ? :-)

Here's my serial output :

_FS not formatted. Doing that now... (can last up to 30 sec).
FS formatted...
---> Now reading time from NTP Server
Current UNIX Timestamp: 1558400803
Time & Date: 1:6:43; 21.5.2019
forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
filter off
Temp: 21.05°C; Humidity: 51.25%; Pressure: 978.49hPa; Pressure rel: 1010hPa; Dewpoint: 10.59°C; 
Not warm enough (less than 26.7 °C) for Heatindex
HeatIndex: 21.05°C; Voltage = 4.66 V
Failed to open file
---> Starting initializing process.
Failed to open file
Stopping process - maybe flash size not set (SPIFFS).

Abort called

>>>stack>>>

ctx: cont
sp: 3ffff8d0 end: 3fffffc0 offset: 01b0
3ffffa80:  3ffef18c 3ffef46c 00000000 4021195a  
3ffffa90:  3ffef18c 3ffef46c 3ffef46c 40202009  
3ffffaa0:  402106d8 00000000 000003e8 40206c45  
3ffffab0:  00000000 00000000 00000000 00000000  
3ffffac0:  3ffef744 7461642f 78742e61 40200074  
3ffffad0:  00000000 00000000 3ffef46c 00000000  
...
3fffff90:  feefeffe feefeffe feefeffe 3ffef588  
3fffffa0:  3fffdad0 00000000 3ffef558 40207aac  
3fffffb0:  feefeffe feefeffe 3ffe8564 40101105  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld_

Upon further investigations, when trying to use the IDE Plugin ESP8266 Sketch Data Upload with Wemos D1 Mini Pro selected as my target device I have an error, and interstingly the auto-detect flash capacity is 4MB. I suspect an issue with my conf (boards.txt ?):

esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
**Auto-detected Flash size: 4MB**
Traceback (most recent call last):
  File "/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2698, in main
    operation_func(esp, args)
  File "/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2125, in write_flash
    % (argfile.name, argfile.tell(), address, flash_end))
esptool.FatalError: File /tmp/arduino_build_469435/Solar-WiFi-Weather-Station-V2.3.spiffs.bin (length 14659584) at offset 2097152 will not fit in 4194304 bytes of flash. **Use --flash-size argument, or change flashing address.**
SPIFFS Upload failed!

Do you have links or ideas on how to investigate this ? The board I have is : https://www.aliexpress.com/item/D1-mini-Pro-16M-bytes-external-antenna-connector-ESP8266-WIFI/32812447045.html
It is specified as having a 16MB Flash

Lefuneste83, 2019-05-21:
Look no further the boards have XMC QH32B memory chips which are 4MB flash modules... Scam...

closed 2019-05-21 by Lefuneste83

newelco, 2019-09-04:
Hi i have the same problem  -
 ...HeatIndex: 21.05°C; Voltage = 4.66 V
Failed to open file
---> Starting initializing process.
Failed to open file
Stopping process - maybe flash size not set (SPIFFS).

Abort called...

Can you have solution for this project with this board ?

Lefuneste83, 2019-09-04:
Check that the actual Flash Memory chip on your ESP board has the proper size by looking at the data sheet for the engraved component number. Mine was undersized to 4MB (D1 Mini non Pro) instead of 16MB (D1 Mini Pro). It was a wrongly specified product by the seller and I got refunded 100% for the boards I purchased. I have since ordered proper Flash modules in 16MB and have replaced them on the ESP boards (you need to know how to solder SMD parts for this). The error message then disappeared.

newelco, 2019-09-06:
you are right - i have the same memory chip - xmc qh32B. i dont have time to wait for new board from china  - im buy in local seller - with winbond chip - and works fine !!!

newelco, 2019-09-06:
can you tell me a seller's name from aliexpress - maybe the same seller - now im start dispute - atach video and wait for finish.

ESF-nl, 2021-03-26:
I think I have a similar issue. The serial monitor runs in a continuous loop: `---> Starting initializing process.
Failed to open file
Stopping process - maybe flash size not set (SPIFFS).
` 
In Arduino IDE I can only select Flash Size: 16 MB. This is right according to the specs I bought https://www.banggood.com/Mini-D1-Pro-Upgraded-Version-of-NodeMcu-Lua-Wifi-Development-Board-Based-on-ESP8266-p-1668897.html?rmmds=myorder&cur_warehouse=CN

Does anyone maybe have any suggestions?

3KUdelta, 2021-03-26:
Check this, maybe it works:
[#39](https://github.com/3KUdelta/Solar_WiFi_Weather_Station/issues/39)

ESF-nl, 2021-03-27:
Great! That helped me. Thank you.