3KUdelta/Solar_WiFi_Weather_Station #2: Error compiling of the 2.3 version on Wemos d1 Pro

keesvw, 2019-06-04:
Hi First of all much respect for your moderation and improvement of the 2.3 version and I would like to get it working also.. The 2.0 model works fine and and get it all working and was also my first IoT project and made me curious to expand it further.
Also I get stuck when compiling your code.

1. I managed to add the adafruits and the blynk libraries but it keeps stalling on not being able to find the espwifi.h library (see screenshot). Perhaps I make a beginner mistake and would appreciate a small push to get me further.
[image]

2. I also added the GY-1145 and add a DS18B20 to monitor a 2nd temperature. I want to make a case modding by adding a window with some perspex so the UV sensor could house there and provide via Blynk a UV warning. I was wondering if you could modify the code  as such that it can read those 2 sensors.

Much appreciate if you could assist me 

Kees

djbriane, 2019-06-11:
I'm also running into the same issue - tried searching for that header file but haven't had any luck. I presume it is part of the ESP8266 library but looking through it's source I can't find it.

djbriane, 2019-06-11:
Solved this by changing `ESPWiFi.h` to `ESP8266WiFi.h` and everything seem to be working fine now.

keesvw, 2019-06-11:
Got that part working now as well; But now it stalls at:
exit status 1
Errot in compil bij het compileren voor board LOLIN(WEMOS) D1 mini Pro

djbriane, 2019-06-12:
I'm not sure about that error, is there anything else in the output? 

Without seeing more all I can suggest is make sure you have all the libraries installed for your board (although I'd assume those would show up in the error log if missing). The other thing I had to do was install the USB drivers for the Wemos D1 (on my Mac) - make sure you've done that as well (though this looks like a compile error rather than a programming issue).

closed 2019-06-28 by 3KUdelta