3KUdelta/Solar_WiFi_Weather_Station #3: MQTT: add support for username and password

rafek00, 2019-07-03:
Hello,
would it be possible to expand MQTT version, with user support and password ?

FrankUlbrich, 2019-07-03:
Yes , it is possible.
I did it yesterday. I can paste the code snipplet if you like .
In my case I added DHT22 sensor as well.

3KUdelta, 2019-07-05:
Thank you Frank.

closed 2019-07-05 by 3KUdelta

FrankUlbrich, 2019-07-06:
3KUdelta - Tkanks a lot for sharing the code.
I added a DHT22 Sensor and would like to transfer data to an INFLUX DB. This will be my next step.
I am waiting for my UV sensor, which I would like to integrate.
Maybe I can share this with you.
Greeting from Germany

3KUdelta, 2019-07-07:
Hi Frank. The easiest way to do this if you just fork my work and put your solution there. I am happy to do a comment on the main page to describe your add-ons.

FrankUlbrich, 2019-07-18:
[Solar-WiFi-Weather-Station-V2.3MQTT-FU.zip](https://github.com/3KUdelta/Solar_WiFi_Weather_Station/files/3407487/Solar-WiFi-Weather-Station-V2.3MQTT-FU.zip)

Hi 3KUdelta,
thank you very much for your offer to fork your briliant solution. In the moment we are going to vacation. When I am back I have to learn how to do it and whcih tool I should use to access gitub for forking. I guess there is a way to sync the code from my laptop.