3KUdelta/Solar_WiFi_Weather_Station #4: Fahrenheit ?

bhamilton1co, 2019-07-27:
Hello, any chance you can add an option to settings.h to change from Celsius to Fahrenheit? Would love to try your project out, but need the readings in Fahrenheit.

bhamilton1co, 2019-07-29:
​I need Fahrenheit for my Temperature readings.​
Do you see any issues with changing this line of code:

From:
`   measured_temp = measured_temp + TEMP_CORR;`

And Changing To:
`   measured_temp = measured_temp *1.8+32;`

It seems to be working. Let me know if you think this will cause any issues, or if you know of a better way to implement Fahrenheit Temperature readings.

Thanks.

3KUdelta, 2019-07-29:
Hi Brian - sorry for the late answer. Your proposal will work without any problems. 
Just in case you want to correct the reading with the TEMP_CORR value then simply use both lines. 
e.g. 
measured_temp = measured_temp + TEMP_CORR;  // correcting the celsius value
measured_temp = measured_temp *1.8+32;             // converting it into Fahrenheit
Marc

closed 2019-07-29 by 3KUdelta

bhamilton1co, 2019-07-29:
Perfect, Thanks!