3KUdelta/Solar_WiFi_Weather_Station #5: Request: implement logging to influxDB for visualizing in Grafana

f4stb00t, 2019-08-08:
Hello 3KUdelta,

thanks a lot for the new code. This is awesome. :)

May I ask for an implementation of new code? I tried it myself now for hundred times, but to be honest my coding skills are less than limited. I like to write small bash scripts, or adjust small things, but with this I am kinda lost. :(

My Goal: Run the Weather Station only local, no internet connection no nothing. Everything runs in a seperated WiFI and VLAN. The Weather Station should connect to the seperated Wifi and send the data of the sensors to my influxDB Server. Finally the data will be visualized via Grafana. I have this running with a small python script with a raspberry and a DHT22. See this: https://www.definit.co.uk/2018/07/monitoring-temperature-and-humidity-with-a-raspberry-pi-3-dht22-sensor-influxdb-and-grafana/

There are already examples for this in the Arduino IDE, means transfering data to influxDB, setting variables and such, but I never got it running, syntax errors or what ever. I am lost :/

Guess with coding experiences it will be that easy for you compared to myself in  tshooting network issues on routers and switches and stuff  in enterprise networks. I am quite sure there are others who would like to run this awesome project only locally withouth any proprietary software.

Kind Regards

fastboot

3KUdelta, 2019-08-08:
Hi fastboot

I hope I understood what you want to do. Unfortunately I am not familiar with connecting to influxDB, nor do I have an environment to test. Maybe someone else here in the community can help you. My 5 cents: Why don't take what is already here and connect to influxDB? 
High level:
- log on to the local network via WiFi (existing code)
- install mosquitto on your pi
- get all the data via MQTT (existing code)
- push MQTT into influxDB (e.g. https://github.com/mhaas/mqtt-to-influxdb-forwarder)

or
- install node-red and use node-red dashboard and forget about influxDB

Hope that helps, Marc

f4stb00t, 2019-08-08:
Hi Marc, 

it seems FrankUlbrich forked your code and implemented the function I was looking for. Just read through the issues and found that he wanted to implement this. Thank you guys! :)

KR

fastboot

3KUdelta, 2019-08-08:
Great news. Thank you. Closing this issue.

closed 2019-08-08 by 3KUdelta