3KUdelta/Solar_WiFi_Weather_Station #9: Virtual pin 10 not logging to Blynk

solbero, 2019-11-05:
The data from virtual pin 10 (dewpoint spread) is not logged to Blynk. It seems like data is not pushed to the Blynk server. All other virtual pins are working fine.

I would have liked to have added the output from the serial monitor to this issue. Unfortunately I have placed the weather station at my cabin, and I will not be going back there for a couple of weeks.

Any suggestions what might be wrong? Could it be a code issue?

3KUdelta, 2019-11-07:
@solbero I double checked the settings and they are ok. Please check your spelling line by line, letter per letter again and also check in Blynk if you have properly assigned the virtual pin 10. Just for testing purposes you could also assign some other free pin.

solbero, 2019-11-07:
Virtual pin 10 is correctly assigned to the Labeled Value widget. Usually it doesn't display any data, but sometimes it displays spurious data (see screenshot).

I'm unsure what you mean by this? Where are you suggesting I check my spelling?

> Please check your spelling line by line, letter per letter again

Here are some screenshot from my Blynk instance. It is cloned from the one referenced in your repository.

[image]

[image]

3KUdelta, 2019-11-08:
When I see your values it could be an error in the algorithm when temperatures go below zero. I´ll check and keep you posted.

3KUdelta, 2019-11-08:
Could you please send the output of serial monitor after a reset?

solbero, 2019-11-08:
I can't give you any serial output at the moment because I have placed the weather station at my cabin. The next time I head to my cabin I'll connect a serial monitor and get some output.

Could this issue just be left open until then?

solbero, 2019-12-01:
Here is the output from the serial monitor after pressing RST button:

```
rl l⸮⸮| ⸮l⸮|⸮l⸮b|⸮⸮⸮⸮r⸮b⸮b⸮⸮nn⸮lnn⸮⸮⸮bp⸮⸮lrlrlp⸮n⸮⸮l⸮⸮bn⸮|l⸮⸮b⸮⸮nn⸮ l⸮⸮l`⸮nnl`nr⸮⸮⸮nb⸮lr⸮⸮nb⸮l⸮⸮ll⸮⸮l`⸮⸮n⸮
Start of SolarWiFiWeatherStation V2.31
---> Connecting to WiFi .... Wifi connected ok
SPIFFS Initialization: (First time run can last up to 30 sec - be patient)
---> Now reading time from NTP Server
Current UNIX Timestamp: 1575198434
Time & Date: 11:7:14; 1.12.2019
forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
filter off
Temp: 16.21°C; Humidity: 70.73%; Pressure: 954.75hPa; Pressure rel: 1015hPa; Dewpoint: 10.89°C; 
Dewpoint Spread: 5.32°C; 
Not warm enough (less than 26.7 °C) for Heatindex
HeatIndex: 16.21°C; Voltage = 3.81 V
---> Now reading from SPIFFS
Timestamp from SPIFFS: 1575198343
Accuracy value read from SPIFFS: 1
Last 12 saved pressure values: 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 1015.00; 
Timestamp difference: 91
---> Now writing to SPIFFS
File written. Now reading file again.
Found in /data.txt = 1575198343
; 1
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 1015.00
; 
---> Calculating Zambretti letter
---> Calculating trend
Current trend: 0.00
Steady
Calculated and rounded Zambretti in numbers: 3.00
This is Zambretti's famous letter: E
********************************************************
Zambretti says: Fine, possibly showers, Steady
Prediction accuracy: 7%
Reason: Not enough weather data yet.
We need 5 hours more to get sufficient data.
********************************************************
Data written to Blink ...
Going to sleep now for 10 Minute(s).
```

solbero, 2019-12-01:
The data from the serial monitor seems fine, but virtual pin 10 is still not displaying on Blynk. There also seems to be missing a space in this line in Solar-WiFi-Weather-Station-V2.3.ino: https://github.com/3KUdelta/Solar_WiFi_Weather_Station/blob/18cc1b926fa005a956d37bac41834e3346f81664/Solar-WiFi-Weather-Station-V2.3.ino#L285

3KUdelta, 2019-12-01:
All good - variable DewPointSpread has the right value. The blank can be neglected. What you can do as a test: Change line 285 to Blynk.virtualWrite(11, DewPointSpread); and change the Virtual Pin in the Blynk App accordingly to 11.

solbero, 2019-12-02:
I'll do that at the first opportunity. I'll be heading back to the cabin in a few weeks time and I'll test your suggestion then.

solbero, 2020-02-24:
I have updated the software to v2.32 and the issue with logging see point to Blynk still persists. When I take the weather station inside and both temperature and dew point are positive values, everything works well. However, when they are negative nothing is logged.

[image]

3KUdelta, 2020-02-24:
Strange. The calc is done in line 420: DewPointSpread = measured_temp - DewpointTemperature;
When DewpontTemperature is negative then there seems to be the problem. The variables allow negative figures, so this shouldn´t be the problem. I struggle with this. I would try to change line 420 like this: DewPointSpread = measured_temp - (DewpointTemperature);
But this is just a wild guess.

fedecatt, 2020-03-02:
I have the same problem
[image]
[image]

3KUdelta, 2020-03-02:
For debugging purposes, what happens if you change Pin9 and Pin10 in the code and in Blynk Mobile?

fedecatt, 2020-03-02:
Exchange pin 9 for pin 10?

3KUdelta, 2020-03-02:
Yes, just for testing. Do the change in the code and on your mobile phone.

fedecatt, 2020-03-02:
Exchange the pins in setting.h and in the Main code, also in blynk but it doesn't work
[image]

fedecatt, 2020-03-02:
stopped transmitting trend now

3KUdelta, 2020-03-03:
OK, thank you for testing. Please, undo all changes.

In the main code it must look like this (from line 306):

 if (App1 == "BLYNK") {
    Blynk.virtualWrite(0, measured_temp);            // virtual pin 0
    Blynk.virtualWrite(1, measured_humi);            // virtual pin 1
    Blynk.virtualWrite(2, measured_pres);            // virtual pin 2
    Blynk.virtualWrite(3, rel_pressure_rounded);     // virtual pin 3
    Blynk.virtualWrite(4, volt);                     // virtual pin 4
    Blynk.virtualWrite(5, DewpointTemperature);      // virtual pin 5
    Blynk.virtualWrite(6, HeatIndex);                // virtual pin 6
    Blynk.virtualWrite(7, ZambrettisWords);          // virtual pin 7
    Blynk.virtualWrite(8, accuracy_in_percent);      // virtual pin 8
    Blynk.virtualWrite(9, trend_in_words);           // virtual pin 9
    Blynk.virtualWrite(10, DewPointSpread);          // virtual pin 10
    Serial.println("Data written to Blink ...");
  }

And on you mobile device in Blynk for Dewpoint "Punto de Rocio" you need to assign it to V5 and leave the values to default: 0 - 1023, set the display to /pin.#/ °C and chose PUSH below.

3KUdelta, 2020-03-03:
@solbero, I think I found it. It is in Blynk settings. You need to tap on the middle of 0 - 1023 and switch off the circle. Then it works.
[image]

fedecatt, 2020-03-03:
> OK, gracias por probar. Por favor, deshaga todos los cambios.
> 
> En el código principal debe verse así (desde la línea 306):
> 
> if (App1 == "BLYNK") {
> Blynk.virtualWrite (0, medido_temp); // pin virtual 0
> Blynk.virtualWrite (1 ,medido_humi); // pin virtual 1
> Blynk.virtualWrite (2 ,medida_pres); // pin virtual 2
> Blynk.virtualWrite (3, rel_pressure_itated); // pin virtual 3
> Blynk.virtualWrite (4, voltios); // pin virtual 4
> Blynk.virtualWrite (5, DewpointTemperature); // pin virtual 5
> Blynk.virtualWrite (6, HeatIndex); // pin virtual 6
> Blynk.virtualWrite (7, ZambrettisWords); // pin virtual 7
> Blynk.virtualWrite (8, precision_in_percent); // pin virtual 8
> Blynk.virtualWrite (9, trend_in_words); // pin virtual 9
> Blynk.virtualWrite (10, DewPointSpread); // pin virtual 10
> Serial.println ("Datos escritos en Blink ...");
> }
> 
> Y en su dispositivo móvil en Blynk para el punto de rocío "Punto de Rocio", debe asignarlo a V5 y dejar los valores predeterminados: 0-1023, configure la pantalla en /pin.#/ ° C y elija PUSH a continuación.

Work, thanks
[image]

3KUdelta, 2020-03-03:
@fedecatt: Me alegro de que haya funcionado ;-)
@solbero : I leave this issue open - you can close it once your'e fine with the solution.

solbero, 2020-03-18:
Logging Dewpoint Spead to Blynk seems to be working now.

closed 2020-03-18 by solbero