3KUdelta/Solar_WiFi_Weather_Station #13: Blynk energy

zangaby, 2019-12-30:
Hello,

Could you please let me know how much more Blynk energy (besides the starting 2000) is needed to load the Blynk project?

Thanks !

3KUdelta, 2020-01-06:
Well I took the 5000 option because I am testing a lot. I would try first with 1000. And then maybe add another 1000 if this was not enough. We are not talking about a lot of money, anyways.I leave the issue open for others to comment.

zangaby, 2020-01-07:
Thank you for your reply !
It would be useful to know how much extra energy is needed as the more energy you buy, the better deal you get.

3KUdelta, 2020-01-16:
Here is my calculation:
2x Superchart = 1800
3x Labeled Display = 1200
2x Gauge = 600

Gives a total of 3'600 Energy needed for Blynk.

closed 2020-01-16 by 3KUdelta