3KUdelta/Solar_WiFi_Weather_Station #18: transmission problems

danielec1980, 2020-02-16:
Hello everyone,
  I uploaded the file:
- Solar-WiFi-Weather-Station-V2.3.ino
- Settings.h
- Translation.h

on the first start the device transmits the data to BLYNK and also to THINGSPEAK. but then it never transmits again and the last sentence comes out in the serial monitor:
13: 13: 49.100 -> Going to sleep now for 1 Minute (s) .rl
but then it doesn't transmit anymore.

why? where can the problem be?

danielec1980, 2020-02-16:
this is the report:
13:18:30.536 -> Start of SolarWiFiWeatherStation V2.32
13:18:30.536 -> Voltage = 4.32 V
13:18:30.604 -> ---> Connecting to WiFi ........... Wifi connected ok
13:18:37.067 -> SPIFFS Initialization: (First time run can last up to 30 sec - be patient)
13:18:37.407 -> ---> Now reading time from NTP Server
13:18:37.642 -> Current UNIX Timestamp: 1581862716
13:18:37.642 -> Time & Date: 14:18:36; 16.2.2020
13:18:37.745 -> forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
13:18:37.745 -> filter off
13:18:37.879 -> Temp: 19.79°C; Humidity: 56.24%; Pressure: 998.71hPa; Pressure rel: 1030hPa; Dewpoint: 10.82°C; 
13:18:37.879 -> Dewpoint Spread: 8.97°C; 
13:18:37.879 -> Not warm enough (less than 26.7 °C) for Heatindex
13:18:37.879 -> HeatIndex: 19.79°C; ---> Now reading from SPIFFS
13:18:37.879 -> Timestamp from SPIFFS: 1581862426
13:18:37.879 -> Accuracy value read from SPIFFS: 2
13:18:37.879 -> Last 12 saved pressure values: 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 1030.00; 
13:18:37.879 -> Timestamp difference: 290
13:18:37.879 -> ---> Now writing to SPIFFS
13:18:37.879 -> File written. Now reading file again.
13:18:37.948 -> Found in /data.txt = 1581862426
; 2
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 1030.00
; 
13:18:37.948 -> ---> Calculating Zambretti letter
13:18:37.948 -> ---> Calculating trend
13:18:37.948 -> Current trend: 0.00
13:18:37.948 -> stabile
13:18:37.948 -> Calculated and rounded Zambretti in numbers: 1.00
13:18:37.948 -> This is Zambretti's famous letter: A
13:18:37.948 -> ********************************************************
13:18:37.948 -> Previsioni meteo: Bel tempo stabile
13:18:37.948 -> Pressione atmosferica: stabile
13:18:37.948 -> Precisione: 15%
13:18:37.948 -> Reason: Not enough weather data yet.
13:18:37.948 -> We need 5 hours more to get sufficient data.
13:18:37.948 -> ********************************************************
13:18:38.601 -> Data written to Blink ...
13:18:38.702 -> Connect to ThingSpeak - OK
13:18:38.702 -> Data written to Thingspeak ...
13:18:38.702 -> INFO: Closing the Wifi connection
13:18:38.771 -> Going to sleep now for 1 Minute(s).rl

3KUdelta, 2020-02-16:
You need to close Jumper 2 JP2 if you are using openenergy's PCB board. 
If you are not using the board you need to connect D0 with RST. Here is a good explanation (https://randomnerdtutorials.com/esp8266-deep-sleep-with-arduino-ide/):
To use timer wake up with ESP8266, you need to connect the RST pin to GPIO 16 which is labeled as D0. The RST pin of the ESP8266 is always HIGH while the ESP8266 is running. However, when the RST pin receives a LOW signal, it restarts the microcontroller. If you set a deep sleep timer with the ESP8266, once the timer ends, GPIO 16 (D0) sends a LOW signal. That means that GPIO 16, when connected to the RST pin, can wake up the ESP8266 after a set period of time.

danielec1980, 2020-02-16:
it is true it works, I closed jp2 and now it works !!
I spent whole days to find the solution!

Thank you very much! you have been very helpful !!

closed 2020-02-16 by danielec1980