3KUdelta/Solar_WiFi_Weather_Station #20: Translate to spanish?

fedecatt, 2020-03-03:
Some help is appreciated

3KUdelta, 2020-03-03:
It would be great if you could add the Spanish translation in the file tranlsation.h

1.) Fork this repository on the top left.
2.) Edit translation.h in your (!) forked environment
3.) Copy and paste an existing block of any language to the end and name it "SP"
4.) Translate the words on the right side e.g. "Wetter wird schön" into "el clima estará bien"
5.) Add a pull request when you are done
6.) Finished - the community says thank you.. ;-)
7.) To use your translation, change in settings.h your language to SP and recompile

fedecatt, 2020-03-04:
#elif LANGUAGE == 'SP'
  const char TEXT_AIR_PRESSURE[]      = "Presión";
  const char TEXT_RISING_FAST[]       = "en rápido aumento";
  const char TEXT_RISING[]            = "en aumento";
  const char TEXT_RISING_SLOW[]       = "en aumento lento";
  const char TEXT_STEADY[]            = "estable";
  const char TEXT_FALLING_SLOW[]      = "en lento descenso";
  const char TEXT_FALLING[]           = "en descenso";
  const char TEXT_FALLING_FAST[]      = "en rápido descenso";
  
  const char TEXT_ZAMBRETTI_FORECAST[]= "Pronóstico";
  const char TEXT_ZAMBRETTI_ACCURACY[]= "Precisión de pronóstico";
  const char TEXT_ZAMBRETTI_A[]       = "Clima estable";
  const char TEXT_ZAMBRETTI_B[]       = "Buen clima";
  const char TEXT_ZAMBRETTI_C[]       = "Mejorando";
  const char TEXT_ZAMBRETTI_D[]       = "Buen clima, pero inestable";
  const char TEXT_ZAMBRETTI_E[]       = "Buen clima con posibles lluvias";
  const char TEXT_ZAMBRETTI_F[]       = "Muy buen clima, mejorando";
  const char TEXT_ZAMBRETTI_G[]       = "Buen clima pero con posibles lloviznas";
  const char TEXT_ZAMBRETTI_H[]       = "Buen clima pero con lloviznas";
  const char TEXT_ZAMBRETTI_I[]       = "Lloviznas pero mejorando";
  const char TEXT_ZAMBRETTI_J[]       = "Variable mejora";
  const char TEXT_ZAMBRETTI_K[]       = "Buen clima con probables lluvias";
  const char TEXT_ZAMBRETTI_L[]       = "Inestable, mejorando después ";
  const char TEXT_ZAMBRETTI_M[]       = "Clima inestable con probables mejoras";
  const char TEXT_ZAMBRETTI_N[]       = "Lluvioso con intervalos";
  const char TEXT_ZAMBRETTI_O[]       = "Lluvioso, se vuelve inestable";
  const char TEXT_ZAMBRETTI_P[]       = "Variable con chubascos";
  const char TEXT_ZAMBRETTI_Q[]       = "Inestable con intervalos cortos de buen tiempo";
  const char TEXT_ZAMBRETTI_R[]       = "Inestable, luego lluvia";
  const char TEXT_ZAMBRETTI_S[]       = "Inestable, lluvia ocasionales";
  const char TEXT_ZAMBRETTI_T[]       = "Muy inestable, pequeños intervalos a veces";
  const char TEXT_ZAMBRETTI_U[]       = "Llueve a veces, luego empeora";
  const char TEXT_ZAMBRETTI_V[]       = "Lluvia ocasional con fuerte inestabilidad";
  const char TEXT_ZAMBRETTI_W[]       = "Precipitaciones frecuentes";
  const char TEXT_ZAMBRETTI_X[]       = "Muy inestable, lluvia";
  const char TEXT_ZAMBRETTI_Y[]       = "Tormenta con posibles mejoras";
  const char TEXT_ZAMBRETTI_Z[]       = "Tormenta, mucha lluvia";
  const char TEXT_ZAMBRETTI_0[]       = "¡Batería baja, por favor recargue!";
  const char TEXT_ZAMBRETTI_DEFAULT[] = "Sin pronóstico por el momento";

3KUdelta, 2020-03-05:
Inserted - thank you.

closed 2020-03-05 by 3KUdelta