3KUdelta/Solar_WiFi_Weather_Station #21: Battery Voltage Monitoring

polanskiman, 2020-03-08:
Hello,

I had a question about the voltage monitoring. Code says:

`//******Battery Voltage Monitoring*********************************************
  
  // Voltage divider R1 = 220k+100k+220k =540k and R2=100k
  float calib_factor = 5.28; // change this value to calibrate the battery voltage
  unsigned long raw = analogRead(A0);
  volt = raw * calib_factor/1024; `

Question is, how one calibrates the battery voltage?

Thank you.

3KUdelta, 2020-03-08:
Hi,

First measure the voltage on your battery. No run the program and you will get the voltage on the serial monitor.  If these two values are different then change the factor (float calib_factor = 5.28;) accordingly.

polanskiman, 2020-03-08:
Hi. Thank you. Not sure exactly how to find the value of the float calib_factor. I'm either getting tired or my brain is not functioning properly.

Voltmeter says 3.983v
Serial monitor say 4.847v (yes I added an additional decimal in the code - lol)

3KUdelta, 2020-03-08:
OK, quick maths:
Formula in the code: volts = raw * calib / 1024
1.) Find raw first: 4.847 = raw * 5.28 / 1024 --> raw = 940
2.) Find calib: 3.983 = 940 * new_calib / 1024 --> your calib_factor: 4.338

Should work...

polanskiman, 2020-03-08:
I am tired indeed!

Thank you.

closed 2020-03-08 by polanskiman