3KUdelta/Solar_WiFi_Weather_Station #22: Battery charging

polanskiman, 2020-04-04:
Hi,

Something strange is happening. Here is a screenshot:

[image]

As you can see I have highlighted in white 3 parts. In the first 2 you can see a voltage spike (on March 9 and around March 16). That's due to me plugging the station to my computer to update the code. After that I placed the station in its usual place where voltage was on a downward trend. That is probably because not enough light or so I thought. However the issue is with the 3rd highlighted part (around March 25 until now). Out of the blue the voltage started spiking on a daily basis. Problem is I haven't moved the station at all. So if the station is actually now properly charging why wasn't it doing it before? The station hasn't been moved from its place and no more or less light is reaching the solar panel.

3KUdelta, 2020-04-04:
Hi
you have a very high voltage overall. If you are operating with a 18650 Li-ion batt then you shouldn´t be over 4.2 if it is fully charged. Normal values for me are around 3.7 V. Your values look strange to me.

You can check the following things:
- Measure the battery with a multimeter. - to compare
- Does the red light go on when the solar panel is pointed to the sun? - check if charging board works ok
- Are you using the board from Open Green Energy? - to be sure about wiring
- If not, can you make sure that the charging board is connected ok? - because high voltage, could be wrong connection
- Did you place a 200k resistor between batt and A0? - because high voltage, could be wrong conversion in code

Happy to get your answers. 
Marc

polanskiman, 2020-04-04:
Hi Marc,

Thank you for the answer. Yes those values are not normal I agree.

- I removed the battery. The multimeters says 3.9v. I would need to test during the day see what the multimeter says while having the battery fitted on the board.

- Yes the red light goes on during the day and off when not enough light.

- Yes the board is the Open Green Energy board. Ordered it from the link pointing to PCBway provided in the instructables.

- Yes the resistor is in place.

Here is a pic of the board:
[image]

polanskiman, 2020-04-05:
I checked the code. The calib_factor was = 5.191v and that's the voltage I was seeing on the serial monitor as well as on the Blynk app. I recalculated the calib_factor. It's 4.049v now.

I am not sure if the new calib_factor that I calculated is going to solve this though. What do you say?

3KUdelta, 2020-04-05:
Hi, the calib_factor is the right thing to tweak. But it is just a mathematical factor, so no v at the end.

// Voltage divider R1 = 220k+100k+220k =540k and R2=100k
  float calib_factor = 5.28; // change this value to calibrate the battery voltage
  unsigned long raw = analogRead(A0);
  volt = raw * calib_factor/1024; 

Here in the last line you see that we take the raw data (0-1024) from the A0 pin and then calculate the respective voltage on the pin. You must find your matching calib_factor to have the same reading in serial monitor as on your multimeter. If this is done, check Blynk. I saw that you are having temp and voltage in the same multidisplay. Maybe delete this on you mobile phone and start from scratch just using virtual pin 4.

[image]

polanskiman, 2020-04-05:
I only added the v here in github. I didn't add that in the code of course.

For the rest that is what I have done. I have recalculate the calib_factor multiple times for some reason something isn't working out right. The battery is clearly charged but the board is reporting under 3.3.

Also it seems that the calib_factor that I am inputing in the code it's the same value given to me from the serial monitor:
[image]
[image]

output.

polanskiman, 2020-04-05:
Well it seems that after fine tuning the calib_factor both the serial monitor and the voltermeter are showing the same thing. Basically I had to recalculate the calib_factor multiple times each time trying to narrow down the figure as accurately as possible.

All this said and done it still does not explain why the voltage started spiking out of nowhere one day without any intervention from me.

polanskiman, 2020-04-13:
I think I spoke too fast. I am having the same issue as before except the peaks are maxed out at 4.56v. That 4.56 is the actual calib_factor value in the code. I am out of ideas. Not sure what’s going on.
[image]

3KUdelta, 2020-04-13:
Hi
this would mean that the analog read is 1024 = max. If I look at your curve it seems that the analog port alters between max and real reading. The upper spike is max and the lower seems to be real reading. There might be something wrong with your analog input. Do you have the possibility to exchange the D1 mini pro?
Marc

3KUdelta, 2020-05-11:
closed, due to inactivity

polanskiman, 2020-05-12:
Due to the Covid situation I have not yet received the new D1 mini pro.

3KUdelta, 2020-05-12:
OK - no worries. You can re-open the issue again if needed. 

Maybe a piece of advice: always order a couple of these D1...

polanskiman, 2020-05-13:
Received the board today. Plugged it. Uploaded the code. Got the following:

`Failed to open file`
`---> Starting initializing process.`
`Failed to open file`
`Stopping process - maybe flash size not set (SPIFFS).`
`Abort called`

bla bla bla

I checked the RAM. Turns out it was a 4MB board. Back to square 1. Seller knows perfectly he tried to scam me and immediately removed the product from sales, gave me a refund (not trough the official channels) and asked me to keep the board.

3KUdelta, 2020-05-13:
Too bad - but you can still use the board for other projects...

polanskiman, 2020-05-13:
Yes. For some unknown reason the problem vanished without any intervention from me for the past 2 weeks but I am sure it will re-appear again.

[image]

3KUdelta, 2020-05-13:
You might to recharge soon. Level on 3.4v. - if reading is correct. I am sure you know to recharge it with USB on the charging board and not on the D1 mini pro.

polanskiman, 2020-05-13:
Yes that's what I have been doing for the past hour. Now at 3.6v. I think I will have to decouple the solar panel from the station so that it can properly charge during the daytime. I don't want the whole station to be exposed to the sun or the temperature readings will be way off.

closed 2020-05-27 by 3KUdelta