3KUdelta/Solar_WiFi_Weather_Station #23: Zambretti algorithm

szerwi, 2020-05-11:
I have a few questions about Zambretti algorithm.
Why there is no possibility to forecast storm when the pressure is falling? I think the pressure is falling before the storm in most cases.
In rising trend case, shouldn't the zambretti be 1 unit less in the winter than in the summer?

3KUdelta, 2020-05-11:
Hi Szerwi, thank you for your comment. You made me dig into the code again - after quite some time. But you are absolutely right with the adjustments over winter and summer. Here is the original source: http://integritext.net/DrKFS/zambretti.htm
"The Forecaster makes an adjustment for Winter or Summer measurements. A Winter falling generally results in a Z value lower by 1 unit compared with a Summer falling pressure. Similarly a Summer rising, improves the prospects by 1 unit over a Winter rising."

So I corrected this right now and generated V2.33. Thank you!

Zambretti forcasts stormy conditions only on very low (steady and rising) pressures. So its theory predicts a storm having very low pressure conditions and a rising (or steady) trend. 

You might be right with your assumption that a strong falling trend might indicate an upcoming storm. I tried to program the original algorithm without any interpretation. Feel free to fork the source and compile your own forecast.

lcfelippe, 2020-05-25:
Hi @3KUdelta,

Reading the Original Source code that you use as reference i have some questions about your code.

The first one is about the equations:

For a falling barometer Zf = 130-0.12P....................................(1)

For a steady barometer Zs = 138 -0.13P.................................(2)

For a rising barometer Zr = 160-0.16P.....................................(3) 

or

For a falling barometer Z = 130-P/81....................................(4)

For a steady barometer Z = 147 -5P/376.................................(5)

For a rising barometer Z = 179-2P/129.....................................(6)

How can you get the equations that you use in your code?

The other question is about Summer and Winter. In Winter falling the Z number is lower by 1 unit compared with summer, why are you decrease the Z number in 1 unit?, shouldn't have to +1? and decrease 1 unit for the Summer Rising?

Thank you in advance.

3KUdelta, 2020-05-26:
Hi, to your 3 points:

1.) Algorithm - I followed the conclusions and these sources:

> Conclusions
To obtain an on-demand weather forecast from an electronic barometer based upon the Zambretti Forecaster, the following steps must be built into the embedded firmware:
A) The determination of the time-derivative of the pressure.
B) On the basis of A), select the appropriate equation (4), (5) or (6)
C) Measure the current pressure
D) Apply the appropriate equation from B) to obtain a value of Z
E) Modify Z according to wind direction data if available,
F) Modify Z according to the Season

Bases for the code learnt from here: 

> http://drkfs.net/zambretti.htm or http://integritext.net/DrKFS/zambretti.htm
  https://raspberrypiandstuff.wordpress.com
  David Bird: https://github.com/G6EJD/ESP32_Weather_Forecaster_TN061

2.) Winter/Sommer adjustments (Point F in conclusion) you made me think again. 

3.) Accuracy is an artificial value - counting the available data points and sum it up to max 94%. The assumption is a maximal accuracy of 94%.

`  int accuracy_in_percent = accuracy*94/12;            // 94% is the max predicion accuracy of Zambretti`

It would be great if you fork the code and challenge it with your own thoughts. 

Marc

3KUdelta, 2020-05-26:
Adjusted Winter/Summer correction. Thanks for the hint.

lcfelippe, 2020-05-26:
@3KUdelta,

I'm working on a Weather Station, and made a try implementing the Zambretti using the (4), (5), (6) equations of my last post. My observation in real-life x Zambretti's Forecast is that Zambretti's prediction in most time is far away from real.

What's your observation in real-life x Zambretti Forecasting, using your code?

Thanks.

3KUdelta, 2020-05-26:
@lcfelippe,

I run this version of the forecaster for more than a year now in Switzerland and I must say that it is surprisingly accurate. Especially if you consult it at 9 am the forecast for the day is very accurate throughout the year.

lcfelippe, 2020-05-26:
@3KUdelta, 

i'll make a try using your equation to see if it will be more accurately. Do you made any other implementation in the code to have the prediction every 9 am?, cause the 9am prediction is the reading of the last 6 hours, right?. How can you manage it?. Sorry about a lot of questions, but  i'm trying to increase the accurate of my prediction, for the next step i'll do the wind derivation to the code. 

Thanks

3KUdelta, 2020-05-26:
Reading through the Zambretti documentation on the web I often came across the prediction would be best around 9am. Yes, this means 6 hours of weighted pressure observation (rising, falling, steady) before. I just happen to control it every morning around nine because I have built a little monitor standing in my office.  (Combining Zambretti, Temperature MQTT messages, OWM 2 day forecast)

[image]
[image]

You could easily store the "Z" letter into SPIFFS every day at 9am and add a button to recall. I don't do it because I like the "rolling" forecast during the day.

Hope this helps, Marc

lcfelippe, 2020-05-26:
@3KUdelta, Thank you for the answers. I made the equations modifications and run it for the prediction.

lcfelippe, 2020-05-26:
@3KUdelta,

I had made some modifications and now it´s more reliable what i observe in real life. I had marked in the picture the information about the Zambretti´s forecast, and include a progress bar. While it´s in test i´ll use the word, when i have it near the accuracy i´ll make some weather icons.

Regards,

[image]

3KUdelta, 2020-05-27:
Very nice, Felippe. I am very much interested in your results.

 Btw. I am currently testing some IAQ sensors. The SGP30 gives the best results by far... Different topic.

lcfelippe, 2020-05-27:
Thank you, I'm using Bosch BME680 for internal, for external I'm using particulate sensor and in future i'll include SO2, Ozone and CO sensor. 
For now, i change the formula and running some tests on Zambretti's Algorithms for best accuracy. Let's wait…

Regards,

closed 2020-07-06 by 3KUdelta

jspanitz, 2020-07-07:
> Thank you, I'm using Bosch BME680 for internal, for external I'm using particulate sensor and in future i'll include SO2, Ozone and CO sensor.

Can you share the code for those sensors and your display and also is that display remote or attached directly to weather station?