3KUdelta/Solar_WiFi_Weather_Station #30: station won't  connect

picard12, 2020-10-02:
I upload the code with my personal wifi and blynk configuration, but It won't connect to my wifi then to the blynk app. I don't know how to find the issue.

3KUdelta, 2020-10-02:
Hello. Unfortunately it is super difficult for us to help you with this little information. Please share the output of the serial monitor then we might have a chance to understand what is going wrong.

picard12, 2020-10-03:
I'm at the beginnig with arduino, sorry. What kind of instuction i need to send to the device?

3KUdelta, 2020-10-03:
Keep you weatherstation plugged in (not on the charger, but on the ESP8266). 
1.) In Arduino open Tools/Serial Monitor
2.) Push the little reset button on the esp8266
3.) Copy the serial monitor output 
4.) Paste it here

picard12, 2020-10-03:
A lot of strange caracter
#⸮bs⸮⸮⸮b⸮n⸮B⸮⸮⸮⸮⸮rGb⸮⸮R⸮bG⸮B#B#⸮⸮⸮m⸮⸮c⸮O⸮+c⸮

3KUdelta, 2020-10-03:
Sorry, I forgot. You have to align the speed of the serial monitor to the definition in the code: `Serial.begin(115200);`.

picard12, 2020-10-03:
Ok now i can see what happen, thanks

22:39:57.956 -> ---> Connecting to WiFi ....................Could not connect to WiFi!
22:40:09.270 -> Going to sleep for 10 minutes and try again.
22:40:09.270 -> INFO: Closing the Wifi connection
22:40:09.370 -> Going to sleep now for 10 Minute(s).
There is some settings wrong.

3KUdelta, 2020-10-03:
Yepp - make sure your wifi credentials in Settings.h are correct.

You only need three files in arduino:
https://github.com/3KUdelta/Solar_WiFi_Weather_Station/blob/master/Solar-WiFi-Weather-Station-V2.3.ino
https://github.com/3KUdelta/Solar_WiFi_Weather_Station/blob/master/Translation.h
https://github.com/3KUdelta/Solar_WiFi_Weather_Station/blob/master/Settings.h

You only need to edit Settings.h. If you are not using ThingSpeak, just do it like this:

`const String App1 = "BLYNK";         // empty string if not applicable -> "" else "BLYNK" 
const String App2 = "";    // empty string if not applicable -> "" else "THINGSPEAK"

/****** Blink or ThingSpeak Settings ****************************************/

char auth[] = "your Blynk Auth Token"; // Blynk Auth Token

char ssid[] = "your SSID";                           // WiFi Router ssid
char pass[] = "your Password";             // WiFi Router password`

picard12, 2020-10-11:
I tried from scratch by setting the indicated parameters a thousand times, but it doesn't connect. What card do I set on blynk? I set up wemos d1 mini and on arduino i set wemos d1 mini pro.

3KUdelta, 2020-10-15:
Ok - do you have a 2.4 GHz Wifi? 
The ESP8266 does not work on 5 GHz.
Maybe you try with your mobile phone - give yourself a hotspot and put the hotspot credentials of your phone to the code.

picard12, 2020-10-17:
Hello, follow your suggestion and i found that my station connect correctly to the hotspot. but won't connect to the modem. Is very strange. My modem is a 2.4Ghz and 5Ghz dual band and I have configured the 2.4GHz network, but without success.

f45tb00t, 2020-10-18:
Hello picard12,

we are really not responsible for network configuration issues. 

In fact what you can do, see the "router" or "access point"  if it offers an IP address via dhcp. You could also change the code of 3KUdelta a bit to show you the ip address.  Example: " Serial.print("Your IP Address is: ");"

Code would be like this:

...
 Serial.print(".");
  }
  Serial.println(" Wifi connected ok");
 Serial.print("Your IP Address is: ");
  Serial.println(WiFi.localIP().toString()); 
...

As you can see you just could paste it between the lines. 
This way at least you can see if you get an ip address.

But I can be routing issues as well. Or even "weird" passwords, which needs to be double quoted. 
What you could test as well is to setup another accesspoint. Give it a straight forward name and especially password to see if the esp is connection correctly. Like password "thisisasupersecurepassword123" and not like Password: !"=)(2342=)"!"""""///=??$%%§$§$....

Cheers

fastboot

picard12, 2020-10-19:
I have a lot of device connect to my modem trought 2.4Ghz channel and all are ok. I'm starting to thing it can be a arduino library issue. My password have only number and letter.

3KUdelta, 2020-10-19:
You could do a clean Arduino installation and re-install all the libraries needed.

picard12, 2020-10-21:
I tryed a lot of time to reset arduino installation, but nothing change. what are the libraries that you have installed and that you know are working?

3KUdelta, 2020-10-21:
As stated in the description of the code:

Needed libraries:
  <Adafruit_Sensor.h>    --> Adafruit unified sensor
  <Adafruit_BME280.h>    --> Adafrout BME280 sensor
  <BlynkSimpleEsp8266.h> --> https://github.com/blynkkk/blynk-library
  <ESPWiFi.h>
  <WiFiUdp.h>
  "FS.h"
  <EasyNTPClient.h>      --> https://github.com/aharshac/EasyNTPClient
  <TimeLib.h>            --> https://github.com/PaulStoffregen/Time.git

For WiFi this is ESPWiFi.h which comes with the ESP8266 distribution. Hopefully you have included in Arduino's settings under "Additional Boards Boards Manager URLs:" the following line: http://arduino.esp8266.com/stable/package_esp8266com_index.json

Also double-check this instructable:  https://www.instructables.com/id/Solar-Powered-WiFi-Weather-Station-V20/

3KUdelta, 2020-10-21:
Sorry, hit the closing button.

picard12, 2020-10-26:
I'm still investigating on this issue. I need more time to find out the solution

picard12, 2020-10-28:
OK. I found the issue. My modem was setting to wifi mixed mode, now work with "N wifi only"

closed 2020-10-28 by picard12