3KUdelta/Solar_WiFi_Weather_Station #36: Blynk and Zambretti

drkoraytuncer, 2020-10-18:
Hi,
How can i see Zambretti forecast at Blynk?
Which widget box should i choose?
Sincerely,

3KUdelta, 2020-10-19:
Hello,
best is to choose "Labeled Value" and set it to Virtual 7.

drkoraytuncer, 2020-10-20:
Thank You. It worked.
BUT: (for example)
At blynk i only see: **Changeable, some rain**
At same tine on Thigspeak i see: **Changeable, some rain. Pressure Steady. Prediction accuracy 94%.**
Is there a reason of these difference?
Sincerely,

3KUdelta, 2020-10-20:
Great! 

Yes, in Blynk we only had over one variable: 
`Blynk.virtualWrite(7, ZambrettisWords);          // virtual pin 7`
in Thingspeak we hand over several variables: 
`postStr+=String(forecast_in_words + ": " + ZambrettisWords + ". " + pressure_in_words + " " + trend_in_words + ". " + accuracy_in_words + " " + accuracy_in_percent + "%25."); // Percentage sign needs to be URL-encoded`

Feel free to add this in your code if needed.

drkoraytuncer, 2020-10-21:
I made two other "labeled values" and i choosed Pins V8 and V9.
It is perfect now.

Congratulations for great coding...

closed 2020-10-21 by drkoraytuncer