3KUdelta/Solar_WiFi_Weather_Station #37: Station stops sometimes

drkoraytuncer, 2020-10-21:
Hi,
I use my station since 2-3 days.
2 days ago it worked very well for ~12 hours and stopped suddenly. It didn't began to send data itself. After ~8 hours i reset it manually, it worked again.
Today after ~14 hours stopped and after ~30 minutes began sending data spontaneously (whitout manual reset) again.
Sleep time is 10 minutes. It send data to Blynk with 10 minutes interwall, but at thingspeak it may change, most often 10 minutes, but sometimes 20-30 minutes.
Any suggestion?

3KUdelta, 2020-10-21:
Hello,
this looks to me like you might have a weak WiFi signal (or some trouble with the Bosch sensor). The station goes to sleep again if it fails to connect to the WiFi. Try it for some time inside and have the USB cable attached to observe the serial monitor.

drkoraytuncer, 2020-10-22:
Hi,
The station is at test period, so it is always inside home, Zero ohm resistance is changed for outer antenna.

I learned that yertesday the internet connection was lost. It seems the station stopped because of this problem.

If the internet connection goes away because of internet provider problem or the modem stops for example because of 220V or adapter problems, will the station stop until manually reset? Can't it be run spontaneously after the provider or modem problem solved?

Sincerely,

3KUdelta, 2020-10-23:
Hello

in fact the station goes to sleep after 20 failed attempts to access the WiFi for 10 minutes and retries it until the battery goes below 3.3v. Then it goes to hibernate to keep the battery alive. Why don't you make yourself familiar with the code?

See here: 

while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    i++;
    if (i > 20) {
      Serial.println("Could not connect to WiFi!");
      Serial.println("Going to sleep for 10 minutes and try again.");
      if (volt > 3.3){
        goToSleep(10);   // go to sleep and retry after 10 min
      }  
      else{
        goToSleep(0);   // hybernate because batt empty
      }
    }

closed 2020-12-13 by 3KUdelta