3KUdelta/Solar_WiFi_Weather_Station #38: Hey there :)

SvensonLF, 2020-12-12:
Today i started with my Weatherstation, but to be honest i run into an Issue...
I always get an Error with the "gotosleep" settings.

Thanks for your Help

/Users/sveneichhorn/Downloads/Solar_WiFi_Weather_Station-master-1/Solar-WiFi-Weather-Station-V2.3/Solar-WiFi-Weather-Station-V2.3.ino: In function 'void setup()':
Solar-WiFi-Weather-Station-V2.3:177:21: error: 'goToSleep' was not declared in this scope
         goToSleep(10);   // go to sleep and retry after 10 min
                     ^
Solar-WiFi-Weather-Station-V2.3:180:20: error: 'goToSleep' was not declared in this scope
         goToSleep(0);   // hybernate because batt empty
                    ^
Solar-WiFi-Weather-Station-V2.3:252:20: error: 'measurementEvent' was not declared in this scope
   measurementEvent();            //get all data from the different sensors
                    ^
Solar-WiFi-Weather-Station-V2.3:256:18: error: 'ReadFromSPIFFS' was not declared in this scope
   ReadFromSPIFFS();              //read stored values and update data if more recent data is available
                  ^
Solar-WiFi-Weather-Station-V2.3:262:18: error: 'FirstTimeRun' was not declared in this scope
     FirstTimeRun();
                  ^
Solar-WiFi-Weather-Station-V2.3:275:36: error: 'WriteToSPIFFS' was not declared in this scope
     WriteToSPIFFS(current_timestamp);                   // update timestamp on storage
                                    ^
Solar-WiFi-Weather-Station-V2.3:278:34: error: 'WriteToSPIFFS' was not declared in this scope
     WriteToSPIFFS(saved_timestamp);                     // do not update timestamp on storage
                                  ^
Solar-WiFi-Weather-Station-V2.3:285:58: error: 'ZambrettiLetter' was not declared in this scope
     ZambrettisWords = ZambrettiSays(char(ZambrettiLetter()));
                                                          ^
Solar-WiFi-Weather-Station-V2.3:285:60: error: 'ZambrettiSays' was not declared in this scope
     ZambrettisWords = ZambrettiSays(char(ZambrettiLetter()));
                                                            ^
Solar-WiFi-Weather-Station-V2.3:291:40: error: 'ZambrettiSays' was not declared in this scope
     ZambrettisWords = ZambrettiSays('0');              // send Message that battery is empty
                                        ^
Solar-WiFi-Weather-Station-V2.3:337:24: error: 'server' was not declared in this scope
     if (client.connect(server,80)) {
                        ^
Solar-WiFi-Weather-Station-V2.3:342:16: error: 'api_key' was not declared in this scope
       postStr+=api_key;   
                ^
/Users/sveneichhorn/Downloads/Solar_WiFi_Weather_Station-master-1/Solar-WiFi-Weather-Station-V2.3/Solar-WiFi-Weather-Station-V2.3.ino: At global scope:
Solar-WiFi-Weather-Station-V2.3:369:3: error: expected unqualified-id before 'if'
   if (volt > 3.3) {          //check if batt still ok, if yes
   ^
Solar-WiFi-Weather-Station-V2.3:372:3: error: expected unqualified-id before 'else'
   else{                      //if not,
   ^
Solar-WiFi-Weather-Station-V2.3:375:1: error: expected declaration before '}' token
 } // end of void setup()
 ^
Mehrere Bibliotheken wurden für "TimeLib.h" gefunden
 Benutzt: /Users/sveneichhorn/Documents/Arduino/libraries/Time-master
 Nicht benutzt: /Users/sveneichhorn/Documents/Arduino/libraries/Time
exit status 1
'goToSleep' was not declared in this scope

3KUdelta, 2020-12-12:
Hi,
yes it seems that your version of the compiler does not do the "normal" auto declaration of the function. This is strange because hundreds of downloads do no complain about this. 
I suggest to: 
- install (or re-install) Arduino IDE 1.8.13
- download again the source code and only change the settings.h, it looks like you also messed up some ; or } in your code
- make sure you have no mess with the libraries (see last message) 
Hope this works.

3KUdelta, 2020-12-12:
Hi again,

probably you only deleted somewhere a { or } --> see these error messages:

/Users/sveneichhorn/Downloads/Solar_WiFi_Weather_Station-master-1/Solar-WiFi-Weather-Station-V2.3/Solar-WiFi-Weather-Station-V2.3.ino: At global scope:
Solar-WiFi-Weather-Station-V2.3:369:3: error: expected unqualified-id before 'if'
if (volt > 3.3) { //check if batt still ok, if yes
^
Solar-WiFi-Weather-Station-V2.3:372:3: error: expected unqualified-id before 'else'
else{ //if not,
^
Solar-WiFi-Weather-Station-V2.3:375:1: error: expected declaration before '}' token
} // end of void setup()
^

So it should be ok to just reload the original source file.

SvensonLF, 2020-12-13:
i reloaded the source file about 10 Times and the Errormessage is always the same...
So okay i will reinstall the Arduino Version and try again. I'm using a Mac maybe thats the reason...
Version 2.0 works without any issue ;)

I will report later if i had any success

Greets from Germany
Sven

SvensonLF, 2020-12-13:
Okay after i installed Arduino on my Windows PC it worked...

3KUdelta, 2020-12-13:
Hi Sven,
I just downloaded the pack myself and tried again. No error at all.
That's what you need to do:
- download the repository as .zip
- add new folder in your sketches environment. e.g.  /Adruino/sketches/Solar-WiFi-Weather-Station-V2.3
- copy only the three files (Solar-WiFi-Weather-Station-V2.3.ino, Translation.h, Settings.h) into this folder
- select in Arduino under Tools/Boards: LOLIN(WEMOS) D1 mini Pro, 80 MHz, Flash, 16M (14M SPIFFS), v2 Lower Memory, Disable, None, Only Sketch, 921600 on /dev/cu.SLAB_USBtoUART (ComPort settings for my MacBook)
- click verify to see if it compiles without errors
- change Settings.h
- click upload to load it on the ESP8266

closed 2020-12-24 by 3KUdelta