3KUdelta/Solar_WiFi_Weather_Station #40: ThingSpeak channel not updated in a regular way

biwizit, 2020-12-26:
Hello,

The ThingSpeak channel is not updated very often with this sketch (less than 10% of the time), even if I have no error at all in in the serial terminal, every ten minutes. It is working much better (almost 100%) with the original code from opengreenenergy, which is weird because it is almost the same piece of code in the two scripts. 

Any idea ? Would it be due to the number of fields which are more important in your sketch (or the additional status) ?

Thank you in advance.

Regards.

biwizit, 2020-12-26:
I confirm that it becomes 100% successfull when I comment out the status lines :

```
// postStr+="&status=";
// postStr+=String(forecast_in_words + ": " + ZambrettisWords + ". " + pressure_in_words + " " + trend_in_words + ". " + accuracy_in_words + " " + accuracy_in_percent + "%25."); // Percentage sign needs to be URL-encoded
```     
I need to investigate why. But the workaround is acceptabe for me at the moment. Thanks.

biwizit, 2020-12-26:
I confirm that everything is fine with status added again but when keeping the translation in english or removing the accents in the french translation.

I imagine that the accents are causing some issues somewhere.

I let this ticket open, in case of you want to make this part of the script more solid.

Thank you for all the job !

Regards.

3KUdelta, 2020-12-28:
Thank you for your posting. https://ww2.mathworks.cn/matlabcentral/cody/groups/14/problems/853 confirms that they do not like strings with french accents. 
Maybe someone likes to re-write the part with ThingSpeak and send chars instead of strings.

danielec1980, 2020-12-29:
congratulations for the code, it's a good job!
I made the metereligca station almost a year ago!
it would be nice to implement with some sensors (for example for wind and rain) and maybe a webcam.

Are you planning to develop any upgrades?
thanks

Daniele