3KUdelta/Solar_WiFi_Weather_Station #42: Cannot verify

ESF-nl, 2021-03-20:
After downloading the Solar Wifi Weather Station I cannot verify the program. This error appears in Arduino IDE:

`invalid conversion from 'char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]`

while this row is marked:
`client.publish("home/weather/solarweatherstation/zletter", tmp1, 1);         // ,1 = retained`

Who can help me please?

3KUdelta, 2021-03-20:
Delete these lines and you´ll be fine.
```
char* tmp1 = &z_letter;
client.publish("home/weather/solarweatherstation/zletter", tmp1, 1);         // ,1 = retained
delay(50);
```
You probably do not need the Z-Letter in MQTT. I did some tests and never deleted it again.

ESF-nl, 2021-03-20:
Thank you for you swift reply! I was really interested in the MQTT version as I'm trying to connect the weather data to my Domoticz :-)

I've deleted these line but than the same error appears here:
`  client.publish("home/weather/solarweatherstation/trend", _trend, 1);      // ,1 = retained`
and if I delete that one as well, the 3rd same error is here: 
`  client.publish("home/weather/solarweatherstation/trendinwords", tmp2, 1);`

Should all that rows also be deleted?

I found also this topic about this error but I do not know how to implement it in your code.
(https://forum.arduino.cc/index.php?topic=586641.0)

3KUdelta, 2021-03-20:
Well, then it is another problem. You should not have to delete all the client.publish lines just because the conversion error.

ESF-nl, 2021-03-20:
Yes, I have "LOLIN(WEMOS) D1 mini Pro" selected

3KUdelta, 2021-03-20:
Strange - I just compiled the code with no errors. 
My environment: Arduino 1.8.13 (Mac)

3KUdelta, 2021-03-20:
Please reload code from Github again, recmpile and copy the full error log.

ESF-nl, 2021-03-20:
Thank you again for your patience helping a beginner.

I used the preferences to also create an expanded log.
[Error log.txt](https://github.com/3KUdelta/Solar_WiFi_Weather_Station/files/6176399/Error.log.txt)
[Error log expanded.txt](https://github.com/3KUdelta/Solar_WiFi_Weather_Station/files/6176400/Error.log.expanded.txt)

3KUdelta, 2021-03-20:
I found this: Bibliotheek pubsubclient-master op versie 1.99.1
Try to update to V 2.8 (actual) --> [https://github.com/knolleary/pubsubclient](url)

ESF-nl, 2021-03-20:
Yes the program has been uploaded into my Wemos! Great you found the issue. 
Now I'm going to add sensors :-)

closed 2021-03-20 by ESF-nl