3KUdelta/Solar_WiFi_Weather_Station #43: deep sleep / wakeup not working

nl2299, 2021-03-29:
First of great piece of software and great extension of the original station.
I used the pcb from opengreen energy. I also added a DHT22 which is not working, always failed to read (DHT11 works perfect) but thats another issue.
Main issue is that the ESP does not wake up again after deep sleep. If i press the reset manually it send again but stops also does not wake up. On the PCB RST and D0 are connected as it should.
`Serial monitor prints: Going to sleep now for 10 Minute(s). `and then no new messages.
I adjusted the script only slightly by removing blynk and thingspeak stuff added mqtt authentication and DHT sensor stuff.

any ideas how to solve this ?

3KUdelta, 2021-03-29:
Hi, sometimes the reset/wakeup does not work when USB is plugged in and jumper (JP2) is on. You might check this.
Can you also measure with a multimeter if RST and D0 (jumper on JP2) are really closed?
Did you change something on the goToSleep() function?

nl2299, 2021-03-29:
sleep function is unchanged. I tested with a multi meter both pins are connected.
After the sleep time the esp blinks (so it wakes up) but nothing is transmitted to the mqtt broker.
Same behavior with unplugged usb.

3KUdelta, 2021-03-29:
So it is not a wake-up problem as stated in the headline. It is a problem in connecting with MQTT?
I cannot help you like this. Please post your serial output.

3KUdelta, 2021-03-29:
Please have a look at Frank´s fork. He has written a version with MQTT and password. [Frank´s Fork](https://github.com/FrankUlbrich/Solar_WiFi_Weather_Station)

nl2299, 2021-03-29:
mqtt is working fine. No mqtt issue as it works after press the reset button. Looks like i need to find a solution by my own.

nl2299, 2021-04-03:
Just for the record: problem was the ESP itself after switching to an lollin wemos d1 mini everything is working.
I just finished my build with custom housing external DHT11, UV sensor and E-ink display :)

jspanitz, 2021-04-03:
> E-ink display :)

please share - would love a local display

nl2299, 2021-04-03:
Once i‘m finally done I will share some details. Maybe I will create a instructable for it Stay tuned