3KUdelta/Solar_WiFi_Weather_Station #44: No sensors found

ESF-nl, 2021-04-20:
After preparing all the hardware (for me the fun part) I started uploading the software (for me the hard part). Initially I got an error on Blynk and therefore decided to turn this option off temporary. Now the Weather Station boots and WiFI and MQTT are connected fine :-)
Unfortunately no sensor has been found. I connected the BME280 and an external DS18B20 for measuring inside my bee hive. The load cells (via HX711) are not connected yet. 
[[image] [image]](url)

 `Start of SolarWiFiWeatherStation V2.34
Voltage = 3.71 V
---> Connecting to WiFi ....... Wifi connected ok
---> Connecting to MQTT, reconnecting MQTT...
Attempting MQTT connection with ESP8266Client-da59connected
MQTT connected ok.
SPIFFS Initialization: (First time run can last up to 30 sec - be patient)
---> Now reading time from NTP Server
Current UNIX Timestamp: 1618949246
Time & Date: 20:7:26; 20.4.2021
Could not find a valid BME280 sensor, check wiring!
forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
filter off
Temp: 0.00°C; Humidity: 0.00%; Pressure: 0.00hPa; Pressure rel: 0hPa; Dewpoint: nan°C; 
Temp adjusted: -1.00°C; Humidity adjusted: nan%; Dewpoint Spread: nan°C; 
Not warm enough (less than 26.7 °C) for Heatindex
HeatIndex: -1.00°C; ---> Now reading from SPIFFS
Timestamp from SPIFFS: 1618947789
Accuracy value read from SPIFFS: 1
Last 12 saved pressure values: 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 
Timestamp difference: 1457
---> Now writing to SPIFFS
File written. Now reading file again.
Found in /data.txt = 1618947789
; 1
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 
---> Calculating Zambretti letter
---> Calculating trend
Current trend: 0.00
stabiel
Calculated and rounded Zambretti in numbers: 138.00
This is Zambretti's famous letter: t
********************************************************
Zambretti says: Sorry, geen voorspellong beschikbaar, stabiel
Prediction accuracy: 7%
Reason: Not enough weather data yet.
We need 5 hours more to get sufficient data.
********************************************************
Connect to ThingSpeak - OK
Data written to Thingspeak ...
INFO: Closing the MQTT connection
INFO: Closing the Wifi connection
Going to sleep now for 10 Minute(s).`

To be sure my sensors are fine (although they are new from the package) I connected the BME280 to another Arduino Uno and used I2C scanner. It indicated I have to use 0x77. Then did the BME280test sketch and the sensor is showing data fine. Then I double checked all my soldering on the Weather Station. I let the Weather Station run for a few days and only the elevation appears in Thingspeak (fixed 'sensor') and all the other sensor still show zero. I don't know what else can be wrong.

Does anybody have a suggestion for me as a beginner?  

Thank you in advance!

PS totally not urgent, but maybe in one of the future updates could you correct a minor spelling mistake in the language file "NL"? "Sorry, geen voorspellong beschikbaar" should be "Sorry, geen voorspelling beschikbaar".

3KUdelta, 2021-04-21:
Hi, if I look at you picture, I believe that not the BME is the problem but the additional DS18B20 temp sensor. As far as I know this sensor is not I2C compatible. So the connection I2C may not be used for this sensor. This might cause the reading errors on the BME. You can use my [instructable](https://www.instructables.com/Crocodile-Solar-Pool-Sensor/) for hooking this sensor to the board. I would then use the connection P2 for this sensor. Please be aware that this sensor also needs some additional resistor. 
Why not first checking everything only with the BME280 and later experiment with an additonal temp sensor?
Hope this helps, Marc

ESF-nl, 2021-04-21:
Thank you very much for your help. I really appreciate it.
I believe the DS18B20 is a 1-wire sensor, but to be sure in eliminating any problem, I removed it. Unfortunately also without the extra temperature sensor, my weather station still said "Could not find a valid BME280 sensor, check wiring!"

`Start of SolarWiFiWeatherStation V2.34
Voltage = 3.72 V
---> Connecting to WiFi ....... Wifi connected ok
---> Connecting to MQTT, reconnecting MQTT...
Attempting MQTT connection with ESP8266Client-5e0fconnected
MQTT connected ok.
SPIFFS Initialization: (First time run can last up to 30 sec - be patient)
---> Now reading time from NTP Server
Current UNIX Timestamp: 1619034745
Time & Date: 19:52:25; 21.4.2021
Could not find a valid BME280 sensor, check wiring!
forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
filter off
Temp: 0.00°C; Humidity: 0.00%; Pressure: 0.00hPa; Pressure rel: 0hPa; Dewpoint: nan°C; 
Temp adjusted: -1.00°C; Humidity adjusted: nan%; Dewpoint Spread: nan°C; 
Not warm enough (less than 26.7 °C) for Heatindex
HeatIndex: -1.00°C; ---> Now reading from SPIFFS
Timestamp from SPIFFS: 1619029291
Accuracy value read from SPIFFS: 1
Last 12 saved pressure values: 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 
Timestamp difference: 5454
---> Now writing to SPIFFS`

Could this problem be related with the installed libraries? I tried many in my attempts to get things running.

3KUdelta, 2021-04-21:
Well, now I spotted the other error. Your Chinese BME board has SCL and SDA mixed up. You need to order a GY-BME280 at Aliexpress or just fix it with jumper wires.
Marc

ESF-nl, 2021-04-21:
Thank you for checking things again. I quickly connected the BME280 with jumper cables and swapped pin 3 and 4. Unfortunately the same persistive error. I've uploaded the code to the Wemos again to be sure.
`Current UNIX Timestamp: 1619046578
Time & Date: 23:9:38; 21.4.2021
Could not find a valid BME280 sensor, check wiring!
forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
filter off
Temp: 0.00°C; Humidity: 0.00%; Pressure: 0.00hPa; Pressure rel: 0hPa; Dewpoint: nan°C; 
Temp adjusted: -1.00°C; Humidity adjusted: nan%; Dewpoint Spread: nan°C; 
Not warm enough (less than 26.7 °C) for Heatindex
HeatIndex: -1.00°C; ---> Now reading from SPIFFS
` 
I'll check for this other sensor as you suggested.

3KUdelta, 2021-04-21:
Too bad, I was sure that was it. 

1. double check the Adafruit libraries. I am running V2.1.2 for BME280
2. double check if the sensor board is still running (not fried yet) - try the Adafruit BME280test from the examples leaving the board plugged in
3. if no success, try to run the example directly hooked to the ESP8266 (not plugged into the green board)
4. if problem i with the green board, use a multimeter to test the connections from D1 (SLC) and D2 (SDA) without power
5. you can also test SLC and SDA with power, both should be high (3.3 V) in running mode (avoid in any case a short)

Let´s see how far you come.

3KUdelta, 2021-04-21:
Hm, try this first. Your board has also the other address, so you need to change this line accordingly:
`  bme_status = bme.begin(0x76);  //address either 0x76 or 0x77`

ESF-nl, 2021-04-24:
Thanks again for your many suggestions. 

0. With the sketch [i2c_scanner](https://gist.github.com/tfeldmann/5411375) I ran on another and Arduino I found that the address should be 0x77 indeed. This was the name I used in your program since the beginning. 
1. I installed library BME280 V2.1.2 as first step. I had installed: cactus_io_BME280_I2C.zip, Adafruit_BME280_Library-master.zip, Core_BME280_i2C.zip and Core_BME280_SPI.zip installed before. No success.
3. Then I measured the voltage. Only on SDI and CSB I measured 3,3V (and VCC of course). These are pins 3 and 5. This could be the area of the error!? I have not common abbreviations on the BME280. The data pins are named: SDI SCK CSB and SDO. So I swapped my pin 3 and 5 so they corresponded with your SLC and SDA as printed on the green board. No success.
2. I removed the BME and tested it on my Arduino Uno again. Same sketch and same wiring as last week (I even took a picture because I was happy the BME test did show data). Now I cannot find any data anymore only "BME test" is shown... then nothing. I'm afraid because all the testing my sensor is not alive any more :-(. I'll order a new set as you suggested before.

Let's close the call. I'll come back with the success story hopefully soon.

closed 2021-04-24 by ESF-nl

ESF-nl, 2021-06-02:
Sorry to bother you again. Finally my new BME280 sensors arrived. Very convinced this suggestion by you would help my weather station. Unfortunately I still receive the message "Could not find a valid BME280 sensor".

`Start of SolarWiFiWeatherStation V2.34
Voltage = 4.75 V
---> Connecting to WiFi . Wifi connected ok
---> Connecting to MQTT, reconnecting MQTT...
Attempting MQTT connection with ESP8266Client-356aconnected
MQTT connected ok.
SPIFFS Initialization: (First time run can last up to 30 sec - be patient)
---> Now reading time from NTP Server
Current UNIX Timestamp: 1622667039
Time & Date: 20:50:39; 2.6.2021
Could not find a valid BME280 sensor, check wiring!
forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
filter off
Temp: 0.00°C; Humidity: 0.00%; Pressure: 0.00hPa; Pressure rel: 0hPa; Dewpoint: nan°C; 
Temp adjusted: -1.00°C; Humidity adjusted: nan%; Dewpoint Spread: nan°C; 
Not warm enough (less than 26.7 °C) for Heatindex
HeatIndex: -1.00°C; ---> Now reading from SPIFFS
Timestamp from SPIFFS: 1622667014
Accuracy value read from SPIFFS: 1
Last 12 saved pressure values: 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 0.00; 
Timestamp difference: 25
---> Now writing to SPIFFS
File written. Now reading file again.
Found in /data.txt = 1622667014
; 1
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 0.00
; 
---> Calculating Zambretti letter
---> Calculating trend
Current trend: 0.00
stabiel
Calculated and rounded Zambretti in numbers: 138.00
This is Zambretti's famous letter: t
********************************************************
Zambretti says: Sorry, geen voorspelling beschikbaar, stabiel
Prediction accuracy: 7%
Reason: Not enough weather data yet.
We need 5 hours more to get sufficient data.
********************************************************
Connect to ThingSpeak - OK
Data written to Thingspeak ...
INFO: Closing the MQTT connection
INFO: Closing the Wifi connection
Going to sleep now for 10 Minute(s).`

Do you maybe have some other suggestions for me as a beginner to try?

Thank you in advance!

3KUdelta, 2021-06-02:
The only error I can see is: Could not find a valid BME280 sensor, check wiring!

- Check version of BME library.
- Check physical wiring with a multimeter for I2C connections. D1 = SLC, D2 = SDA