3KUdelta/Solar_WiFi_Weather_Station #45: Temperature offset vs thingspeak

Spr1DLy, 2021-05-05:
Hi.
Have a problem. Weather station upload temperature without offset parameter (my bme280 made measurements +5 degree).

Pls hlp!

3KUdelta, 2021-05-05:
You are absolutely right. Thank you for spotting this. 

Change the lines (346)

```
      postStr+=String(measured_temp);
      postStr+="&field3=";
      postStr+=String(measured_humi);
```
into
```
      postStr+=String(adjusted_temp);
      postStr+="&field3=";
      postStr+=String(adjusted_humi);
```
or simply download the code again. It is corrected now.

Spr1DLy, 2021-05-10:
Thx a lot. It works!

Another question came up. The rel_press value significantly exceeds the actual pressure values, and the press value has the correct pressure. what exactly does rel_press show?

Here in Russia we use mmHg instead hPa, so 991.82 hPa = 743.92823389988 mmHg (corresponds to the official forecast)

16:24:46.924 -> Temp: 25.25°C; Humidity: 18.63%; Pressure: 991.82hPa; Pressure rel: 1053hPa; Dewpoint: -0.31°C; 
16:24:46.924 -> Temp adjusted: 20.75°C; Humidity adjusted: 24.44%; Dewpoint Spread: 25.56°C; 

PS by the way, I made a translation into Russian, I can share it if you want.

3KUdelta, 2021-05-10:
Hi

For the difference of the two please see [here](https://help.ambientweather.net/help/what-is-the-difference-between-absolute-and-relative-pressure/). For weather forecast they use relative pressure.

In Settings.h you need to set your location's altitude. I live in Switzerland so my location is 505m ASL. Just change the value for your place in Settings.h. 
`#define ELEVATION (505)             //Enter your elevation in m ASL to calculate rel pressure (ASL/QNH) at your place`

For the Russian conversion you need to do the following:
- add a new variable (add line 125) `float rel_pressure_RUS;` 
- add conversion (add line 415) `rel_pressure_RUS = rel_pressure_rounded / 133.33;`
- send new variable to Blynk (change line 320)   `Blynk.virtualWrite(3, rel_pressure_RUS);     // virtual pin 3`
- send new variable to ThingSpeak (change line 344) `postStr+=String(rel_pressure_RUS);`

I am very happy to post the Russian translation for other Russian users. You can do a pull request here in github for Translation.h or you just send it here in the comments.

closed 2021-05-22 by 3KUdelta