3KUdelta/Solar_WiFi_Weather_Station #48: ThingSpeak not updating anymore

polanskiman, 2022-08-24:
Hello,
Due to some update done on ThingSpeak end, the code is broken and data is being sent anymore to ThingSpeak server. It is discussed here: https://www.mathworks.com/matlabcentral/discussions/thingspeak/745015-all-my-thingspeak-channels-stopped-receiving-data-on-the-same-date-april-13th-2022-at-around-10-00?s_tid=srchtitle

Since I do't often check my ThingSpeak channel, I actually noticed a few days ago that my ThingSpeak Channel also stopped receiving data at the exact same date (April 13). Having little to no knowledge on coding I can't fix this myself.

Any help would be appreciated.

Thank you.

Maigoland, 2022-08-25:
Hi
I have the same problem, the last data published in ThingSpeak is on 2022/12/4.

Any ideas to solve this problem?

Thanks

3KUdelta, 2022-08-25:
It seems that they have changed parts of the file format. I did not notice it either. Probably not a big deal - I will have a look into it.

polanskiman, 2022-08-26:
Seems to be an url encoding issue. Spaces should be formated as %20
Here it is disscussed too: https://www.mathworks.com/matlabcentral/discussions/thingspeak/744635-channel-status-update-strings-including-space-chars?s_tid=srchtitle

3KUdelta, 2022-08-26:
Do you want to try to fix it? I can start looking into it coming Sunday.

polanskiman, 2022-08-26:
I would if I had the knowledge. It can wait. It’s been down for 5 months, a few more days won’t change anything ;)

polanskiman, 2022-08-28:
If I may suggest, in addition to the fix above, it might be worth adding some piece of code to know what happened when a requests was sent to the server. As it currently is, there is no feedback and nothing is printed in the serial monitor as to wether the request was successful or not. We get a "Data written to Thingspeak ..." but that happens no matter what, so it's not very useful to track potential problems.

3KUdelta, 2022-08-29:
Good idea - will look into it.

polanskiman, 2022-08-30:
Wanting to find the solution I contacted ThingSpeak directly. So far I haven't found the way, however at ThingSpeak they recommended using the ThingSpeak library for arduino and esp instead.

https://www.mathworks.com/matlabcentral/answers/1789110-thingspeak-channel-stopped-receiving-data-since-april-13th-2022

3KUdelta, 2022-08-30:
Fixed it with by using the ThingSpeak library. Please be aware that you need to enter your channel id and your key in Settings.h.

polanskiman, 2022-08-30:
Awesome. Will try it out.

polanskiman, 2022-08-30:
All is working smoothly for me now. Also thank you for pointing out the Blynk code will be deprecated by the end of the year. I will try to look into how to update it. Gives me 4 months to learn about coding ;)

closed 2022-08-31 by polanskiman

jspanitz, 2022-08-31:
> Blynk code will be deprecated by the end of the year

Hmm, where is this noted and what does it mean in terms of using the existing blynk app - will it stop working or just be unsupported?

3KUdelta, 2022-08-31:
Not quite true - there are so many replacements:
- ThingSpeak Clients for Android and iPhone (working)
- MQTT (working)
- Arduino IoT Remote (needs to be integrated)
- New Blynk (has also some free version - needs to be integrated)

polanskiman, 2022-09-04:
@jspanitz I have updated the code in my repo to work with the New Blynk. Have a look at it. Changes are really minimal.

3KUdelta, 2023-01-07:
I finally found some time  - thank you @polanskiman for your input. I also have done the few updates for Blynk. It is more to do on Blynk's side than on the code here.