3KUdelta/Solar_WiFi_Weather_Station #49: Add login/pass for MQTT

TomaszDom, 2023-01-21:
My MQTT server requires a login and password, is there an easy way to add this feature?

3KUdelta, 2023-01-21:
Hi, this is pretty simple:

In settings.h you must have these three lines
const char* mqtt_server = "";       // MQTT Server (broker) address
const char* mqtt_user = "";         // MQTT Server (broker) userid
const char* mqtt_pass = "";         // MQTT Server (broker) password

Then in the main code exchange line 795 with
if (client.connect(clientId.c_str(), mqtt_user, mqtt_pass)) {

That's it. Marc

TomaszDom, 2023-01-21:
That makes sense, although the MQTT version of the project does not compile (with or without any changes):

```
E:\Solar_WiFi_Weather_Station\Solar-WiFi-Weather-Station-V2.3 MQTT\Solar-WiFi-Weather-Station-V2.3 MQTT.ino: In function 'void setup()':
E:\Solar_WiFi_Weather_Station\Solar-WiFi-Weather-Station-V2.3 MQTT\Solar-WiFi-Weather-Station-V2.3 MQTT.ino:179:22: error: cannot convert 'PubSubClient' to 'Client&'
  179 |     ThingSpeak.begin(client);  // Initialize ThingSpeak
      |                      ^~~~~~
      |                      |
      |                      PubSubClient
In file included from E:\Solar_WiFi_Weather_Station\Solar-WiFi-Weather-Station-V2.3 MQTT\Solar-WiFi-Weather-Station-V2.3 MQTT.ino:95:
c:\Users\PC\Documents\Arduino\libraries\ThingSpeak\src/ThingSpeak.h:116:29: note:   initializing argument 1 of 'bool ThingSpeakClass::begin(Client&)'
  116 |         bool begin(Client & client)
      |                    ~~~~~~~~~^~~~~~

exit status 1

Compilation error: cannot convert 'PubSubClient' to 'Client&'`
```
I can compile the non-MQTT version without any issues.

3KUdelta, 2023-01-21:
Just to make sure, you have PubSubClient by Nick O'Leary Version 2.8.0 installed?

TomaszDom, 2023-01-21:
Correct.

[image]

3KUdelta, 2023-01-21:
try using on line 179 

ThingSpeak.begin(espClient);

TomaszDom, 2023-01-21:
It worked. :)

3KUdelta, 2023-01-21:
Perfect. Let me know if the MQTT also works.

TomaszDom, 2023-01-21:
It does connect, but it only publishes the zletter.

[image]

3KUdelta, 2023-01-21:
For me to understand - do you also get a final "SolarWeatherstation: Taking a nap for.." message under debug?

3KUdelta, 2023-01-21:
If possible, please list the serial output.

TomaszDom, 2023-01-21:
Yes, it takes 10 minute naps, everything is working as it should for Blynk and ThingSpeak, just not for MQTT, where the debug message also gets updated every 10 minutes. I'll get serial output soon.

TomaszDom, 2023-01-21:
```
....... Wifi connected ok
---> Connecting to MQTT, reconnecting MQTT...
Attempting MQTT connection with ESP8266Client-3b70connected
MQTT connected ok.
SPIFFS Initialization: (First time run can last up to 30 sec - be patient)
---> Now reading time from NTP Server
Current UNIX Timestamp: 1674346965
Time & Date: 0:22:45; 22.1.2023
forced mode, 1x temperature / 1x humidity / 1x pressure oversampling,
filter off
Temp: 24.67°C; Humidity: 39.37%; Pressure: 1028.72hPa; Pressure rel: 1029hPa; Dewpoint: 9.92°C; 
Temp adjusted: 23.67°C; Humidity adjusted: 41.80%; Dewpoint Spread: 13.75°C; 
Not warm enough (less than 26.7 °C) for Heatindex
HeatIndex: 23.67°C; ---> Now reading from SPIFFS
Timestamp from SPIFFS: 1674345816
Accuracy value read from SPIFFS: 12
Last 12 saved pressure values: 1029.00; 1029.00; 1028.00; 1028.00; 1027.00; 1027.00; 1027.00; 1026.00; 1026.00; 1025.00; 1025.00; 1025.00; 
Timestamp difference: 1149
---> Now writing to SPIFFS
File written. Now reading file again.
Found in /data.txt = 1674345816
; 12
; 1029.00
; 1029.00
; 1028.00
; 1028.00
; 1027.00
; 1027.00
; 1027.00
; 1026.00
; 1026.00
; 1025.00
; 1025.00
; 1025.00
; 
---> Calculating Zambretti letter
---> Calculating trend
Current trend: 0.74
powoli rośnie
Calculated and rounded Zambretti in numbers: 1.00
This is Zambretti's famous letter: A
********************************************************
Zambretti says: Dobra pogoda, stabilna, powoli rośnie
Prediction accuracy: 94%
********************************************************
Data written to Blink ...
Writing to Thingspeak was successful!
INFO: Closing the MQTT connection
INFO: Closing the Wifi connection
Going to sleep now for 10 Minute(s).

3KUdelta, 2023-01-21:
I am not sure if you answered this - do you also get a final "SolarWeatherstation: Taking a nap for.." message under debug?

TomaszDom, 2023-01-21:
Sorry I don't see that message. Only Sensor started, Client started, every time.

3KUdelta, 2023-01-21:
Ok, this means that the MQTT server disconnects after 3 messages:
1. message:
184 client.publish("home/debug", "SolarWeatherstation: Sensor started");
2. message:
798 client.publish("home/debug", "SolarWeatherstation: client started...");
3. message:
653 client.publish("home/weather/solarweatherstation/zletter", tmp1, 1);

and then stops publishing which makes no sense. Next message would be:
353 client.publish("home/weather/solarweatherstation/tempc", _adjusted_temp, 1);

Right now I believe that there might be a timeout on your MQTT server.

TomaszDom, 2023-01-21:
Strange. It's already handling hundreds of messages to/from Home Assistant and zigbee2mqtt without any problems. I really don't want to use Blynks rest api to get the weather into HA if I can do this directly through mqtt. 

Not sure what to do here.

TomaszDom, 2023-01-21:
[image]
```
We should see this message if the program reaches this place...
0-3INFO: Closing the MQTT connection
```
From documentation:
```
-3 : MQTT_CONNECTION_LOST - the network connection was broken
```
Why though?

3KUdelta, 2023-01-21:
you can try 

connect_to_MQTT(); 

at this point.

TomaszDom, 2023-01-21:
That worked :)

3KUdelta, 2023-01-21:
Still strange why it disconnects there. But you can leave it like this - does not harm.

TomaszDom, 2023-01-21:
I will, thanks for you support! 

FWIW the mqtt server I'm running is:<html><body>

mosquitto-nossl | 2.0.15-1
-- | --

</body>
</html>
on latest OpenWRT, with no special settings other than login/pass req.

closed 2023-01-22 by 3KUdelta