3KUdelta/Solar_WiFi_Weather_Station #52: Suggestion: modifying the TP4056 to operate closer to the MPP of the panel

FBMinis, 2023-09-25:
It is explained here that these little 5V panels will never deliver the 1A current the TP4056 is setup to pull:

https://youtu.be/37kGva3NW8w?t=320

I have determined that the MPP of my solar panel under full sunlight is reached with current around 80mA using this test method:

https://makersportal.com/blog/solar-panel-characterization-and-experiments-with-arduino
https://wolles-elektronikkiste.de/en/ina219-current-and-power-sensor

I suggest replacing the 1.2K resistor on the TP4056 with a 20K trimmer potentiometer and slowly increase the resistance until the power value reaches a maximum.

[Simple_method_for_I-V_characterization_curve_for_l.pdf](https://github.com/3KUdelta/Solar_WiFi_Weather_Station/files/12718514/Simple_method_for_I-V_characterization_curve_for_l.pdf)

[Sunlight based I-V characterization of solar PV.pdf](https://github.com/3KUdelta/Solar_WiFi_Weather_Station/files/12718515/Sunlight.based.I-V.characterization.of.solar.PV.pdf)

Thank you for your project, I'm going to try to build it. The code is very well commented, I think it will be straightforward.

Regards

3KUdelta, 2023-09-26:
Thank you for your valuable input. My stations have been running flawlessly for several years now and the battery keeps charging perfectly. I am curious about your result. 
Marc

FBMinis, 2023-09-26:
That is motivating to know, thank you!

Regarding the BME280, have you had any trouble reading humidity values? I have had one outside for 3 months now and it started reading 100% most of the day.  I searched online and many people complain about this.

3KUdelta, 2023-09-26:
Yes, it needs to be recalibrated once a while. In these cases I take the station to the inside for 24 hours and this will do the trick. However, I struggled much more with the BME's temperature readings that's why I added an extra temp sensor.

FBMinis, 2023-10-19:
Regarding the BME280 stuck to 100% RH, I left it inside a small container covered with isopropanol for a few hours, dried it with a paper towel and then with a blow dryer (as hot as I could). The readings are now in line with other sensors I had around, we'll see for how long.

I'm running your code on an ESP32-CAM that I have outside facing the horizon. Regarding SPIFFS frequent read/write, have you noticed any degradation of your units?

3KUdelta, 2023-10-22:
Thank you for your feedback - very interesting build. Maybe you want to share one day your code in your fork of this project. After all these years outside I never was facing any problems nor with SPIFFS (besides of the case printed in PLA which kind of melted this year).
Marc

3KUdelta, 2024-10-21:
> Regarding SPIFFS frequent read/write, have you noticed any degradation of your units?

Please see actual version, where MQTT is used instead of SPIFFS for storage historical values.
Marc

closed 2024-10-21 by 3KUdelta