3KUdelta/Solar_WiFi_Weather_Station #53: Compilation error: Settings_mst.h: No such file or directory

maba64, 2023-10-28:
I get this error
fC:\Users\mabab\Downloads\Solar_WiFi_Weather_Station-master (4)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:95:10: fatal error: Settings_mst.h: No such file or directory
   95 | #include "Settings_mst.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: Settings_mst.h: No such file or directory
Markus

3KUdelta, 2023-10-28:
Hi Markus,

thank you for mention this - my bad. I corrected the file. It should be Settings.h on line 95.

Marc

maba64, 2023-10-28:
hi Mark
but i get a new error....
thanks

3KUdelta, 2023-10-28:
What kind of error? 
Did you copy the Settings.h into the same folder as the .ino File? Did you fill in all the necessary settings?
Marc

closed 2023-10-28 by 3KUdelta