3KUdelta/Solar_WiFi_Weather_Station #54: new error

maba64, 2023-10-28:
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino: In function 'void setup()':
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:165:3: error: 'translate' was not declared in this scope
  165 |   translate();                      // assign translation variables
      |   ^~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:313:25: error: 'TEXT_ZAMBRETTI_FORECAST' was not declared in this scope
  313 |     forecast_in_words = TEXT_ZAMBRETTI_FORECAST;
      |                         ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:314:25: error: 'TEXT_AIR_PRESSURE' was not declared in this scope
  314 |     pressure_in_words = TEXT_AIR_PRESSURE;
      |                         ^~~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:315:25: error: 'TEXT_ZAMBRETTI_ACCURACY' was not declared in this scope
  315 |     accuracy_in_words = TEXT_ZAMBRETTI_ACCURACY;
      |                         ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino: In function 'void measurementEvent()':
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:501:51: error: 'TEXT_STORM_LOW' was not declared in this scope
  501 |   if (rel_pressure_rounded < 990) PressureState = TEXT_STORM_LOW;
      |                                                   ^~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:502:88: error: 'TEXT_STRONG_LOW' was not declared in this scope
  502 |   else if (rel_pressure_rounded >= 990 && rel_pressure_rounded < 1000) PressureState = TEXT_STRONG_LOW;
      |                                                                                        ^~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:503:89: error: 'TEXT_LOWPRESSURE' was not declared in this scope
  503 |   else if (rel_pressure_rounded >= 1000 && rel_pressure_rounded < 1013) PressureState = TEXT_LOWPRESSURE;
      |                                                                                         ^~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:504:89: error: 'TEXT_HIGHPRESSURE' was not declared in this scope
  504 |   else if (rel_pressure_rounded >= 1013 && rel_pressure_rounded < 1025) PressureState = TEXT_HIGHPRESSURE;
      |                                                                                         ^~~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:505:58: error: 'TEXT_STRONG_HIGH' was not declared in this scope
  505 |   else if (rel_pressure_rounded >= 1025) PressureState = TEXT_STRONG_HIGH;
      |                                                          ^~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino: In function 'int CalculateTrend()':
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:546:22: error: 'TEXT_RISING_FAST' was not declared in this scope
  546 |     trend_in_words = TEXT_RISING_FAST;
      |                      ^~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:550:22: error: 'TEXT_RISING' was not declared in this scope
  550 |     trend_in_words = TEXT_RISING;
      |                      ^~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:554:22: error: 'TEXT_RISING_SLOW' was not declared in this scope
  554 |     trend_in_words = TEXT_RISING_SLOW;
      |                      ^~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:558:22: error: 'TEXT_STEADY' was not declared in this scope
  558 |     trend_in_words = TEXT_STEADY;
      |                      ^~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:562:22: error: 'TEXT_FALLING_SLOW' was not declared in this scope
  562 |     trend_in_words = TEXT_FALLING_SLOW;
      |                      ^~~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:566:22: error: 'TEXT_FALLING' was not declared in this scope
  566 |     trend_in_words = TEXT_FALLING;
      |                      ^~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:570:22: error: 'TEXT_FALLING_FAST' was not declared in this scope
  570 |     trend_in_words = TEXT_FALLING_FAST;
      |                      ^~~~~~~~~~~~~~~~~
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino: In function 'String ZambrettiSays(char)':
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:657:3: error: 'translate' was not declared in this scope
  657 |   translate();
      |   ^~~~~~~~~
[... cut ...]
C:\Users\mabab\OneDrive\Desktop\arduino-ide_2.2.1_Windows_64bit\Solar-WiFi-Weather-Station-V2.4_001\Solar_WiFi_Weather_Station-master (5)\Solar_WiFi_Weather_Station-master\Solar-WiFi-Weather-Station-V2.4\Solar-WiFi-Weather-Station-V2.4.ino:687:34: error: 'TEXT_ZAMBRETTI_DEFAULT' was not declared in this scope
  687 |     default:  zambrettis_words = TEXT_ZAMBRETTI_DEFAULT; break;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: 'translate' was not declared in this scope

3KUdelta, 2023-10-28:
Very well - you need three files in the same directory to make this thing fly:
Solar-WiFi-Weather-Station-V2.4.ino
Settings.h
Translation.h

Where in Settings.h you fill in all your credentials.
Marc

maba64, 2023-10-30:
hi marc
yes they are on the same directory ?
br Markus

maba64, 2023-10-31:
Hi Mark I get this error ?  Gesendet von Mail für Windows Von: MarcGesendet: Samstag, 28. Oktober 2023 18:05An: 3KUdelta/Solar_WiFi_Weather_StationCc: maba64; AuthorBetreff: Re: [3KUdelta/Solar_WiFi_Weather_Station] new error (Issue #54) Very well - you need three files in the same directory to make this thing fly:Solar-WiFi-Weather-Station-V2.4.inoSettings.hTranslation.hWhere in Settings.h you fill in all your credentials.Marc

3KUdelta, 2023-10-31:
Markus,
this error does not come from the sketch - I have no idea what program and where you got this. It seems to me that you have a mess with your directory structure. Best would be you start all over with the default settings and directories.

1. save your sketches to a different location e.g. Desktop
2. save your libraries to a different location e.g. Desktop
3. delete the Arduino directory
4. re-install Arduino with the latest release (follow the instructions)
5. copy your saved sketches back to the /sketches directory
6. copy your saved libraries back to the /libraries directory

Again: "In all supported OSes, there's a directory called Arduino that contains a /hardware, /libraries. and /sketches directory. On Windows, it's in C:/user/XXXXX/Documents/Arduino where XXXXX is the username."
Best Marc

maba64, 2023-11-01:
Hi Marc

I have  made e new installation on a other PC…. Same result…

 

Compilation error: 'translate' was not declared in this scope

maba64, 2023-11-01:
/*----------------------------------------------------------------------------------------------------
  Project Name : Solar Powered WiFi Weather Station V2.4
  Features: temperature, dewpoint, dewpoint spread, heat index, humidity, absolute pressure, relative pressure, battery status and
  the famous Zambretti Forecaster (multi lingual)
  Authors: Keith Hungerford, Debasish Dutta and Marc Stähli
  Website : www.opengreenenergy.com
  
******* Transaltion tables ****************************************/

#if LANGUAGE == 'EN'
  const char TEXT_AIR_PRESSURE[]      = "Pressure";
  const char TEXT_RISING_FAST[]       = "Rising fast";
  const char TEXT_RISING[]            = "Rising";
  const char TEXT_RISING_SLOW[]       = "Rising slow";
  const char TEXT_STEADY[]            = "Steady";
  const char TEXT_FALLING_SLOW[]      = "Falling slow";
  const char TEXT_FALLING[]           = "Falling";
  const char TEXT_FALLING_FAST[]      = "Falling fast";
  
  const char TEXT_ZAMBRETTI_FORECAST[]= "Forecast";
  const char TEXT_ZAMBRETTI_ACCURACY[]= "Prediction accuracy";
  const char TEXT_ZAMBRETTI_A[]       = "Settled fine weather";
  const char TEXT_ZAMBRETTI_B[]       = "Fine weather";
  const char TEXT_ZAMBRETTI_C[]       = "Becoming fine";
  const char TEXT_ZAMBRETTI_D[]       = "Fine, becoming less settled";
  const char TEXT_ZAMBRETTI_E[]       = "Fine, possibly showers";
  const char TEXT_ZAMBRETTI_F[]       = "Fairly fine, improving";
  const char TEXT_ZAMBRETTI_G[]       = "Fairly fine, possibly showers early";
  const char TEXT_ZAMBRETTI_H[]       = "Fairly fine, showers later";
  const char TEXT_ZAMBRETTI_I[]       = "Showery early, improving";
  const char TEXT_ZAMBRETTI_J[]       = "Changeable, improving";
  const char TEXT_ZAMBRETTI_K[]       = "Fairly fine, showers likely";
  const char TEXT_ZAMBRETTI_L[]       = "Rather unsettled, clearing later";
  const char TEXT_ZAMBRETTI_M[]       = "Unsettled, probably improving";
  const char TEXT_ZAMBRETTI_N[]       = "Showery, bright intervals";
  const char TEXT_ZAMBRETTI_O[]       = "Showery, becoming unsettled";
  const char TEXT_ZAMBRETTI_P[]       = "Changeable, some rain";
  const char TEXT_ZAMBRETTI_Q[]       = "Unsettled, short fine intervals";
  const char TEXT_ZAMBRETTI_R[]       = "Unsettled, rain later";
  const char TEXT_ZAMBRETTI_S[]       = "Unsettled, rain at times";
  const char TEXT_ZAMBRETTI_T[]       = "Very unsettled, finer at times";
  const char TEXT_ZAMBRETTI_U[]       = "Rain at times, worse later";
  const char TEXT_ZAMBRETTI_V[]       = "Rain at times, becoming very unsettled";
  const char TEXT_ZAMBRETTI_W[]       = "Rain at frequent intervals";
  const char TEXT_ZAMBRETTI_X[]       = "Very unsettled, rain";
  const char TEXT_ZAMBRETTI_Y[]       = "Stormy, possibly improving";
  const char TEXT_ZAMBRETTI_Z[]       = "Stormy, much rain";
  const char TEXT_ZAMBRETTI_0[]       = "Battery empty, please recharge!";
  const char TEXT_ZAMBRETTI_DEFAULT[] = "Sorry, no forecast at the moment";

#elif LANGUAGE == 'DE'
  const char TEXT_AIR_PRESSURE[]      = "Luftdruck";
  const char TEXT_RISING_FAST[]       = "rasch steigend";
  const char TEXT_RISING[]            = "steigend";
  const char TEXT_RISING_SLOW[]       = "langsam steigend";
  const char TEXT_STEADY[]            = "beständig";
  const char TEXT_FALLING_SLOW[]      = "langsam fallend";
  const char TEXT_FALLING[]           = "fallend";
  const char TEXT_FALLING_FAST[]      = "rasch fallend";
  
  const char TEXT_ZAMBRETTI_FORECAST[]= "Wettervorhersage";
  const char TEXT_ZAMBRETTI_ACCURACY[]= "Vorhersagegenauigkeit";
  const char TEXT_ZAMBRETTI_A[]       = "Beständiges Schönwetter";
  const char TEXT_ZAMBRETTI_B[]       = "Schönes Wetter";
  const char TEXT_ZAMBRETTI_C[]       = "Wetter wird gut";
  const char TEXT_ZAMBRETTI_D[]       = "Schön, wird wechselhaft";
  const char TEXT_ZAMBRETTI_E[]       = "Schön, Regenschauer möglich";
  const char TEXT_ZAMBRETTI_F[]       = "Ziemlich gut, verbessert sich";
  const char TEXT_ZAMBRETTI_G[]       = "Ziemlich gut, frühe Regenschauer möglich";
  const char TEXT_ZAMBRETTI_H[]       = "Ziemlich gut, spätere Regenschauer";
[... 324 lines cut ...]
  const char TEXT_ZAMBRETTI_E[]       = "Pent, mulighet for lette byger";
  const char TEXT_ZAMBRETTI_F[]       = "Stort sett oppholdsvær, senere pent";
  const char TEXT_ZAMBRETTI_G[]       = "Stort sett oppholdsvær, mulighet for tidlige lette byger";
  const char TEXT_ZAMBRETTI_H[]       = "Stort sett oppholdsvær, senere lette byger";
  const char TEXT_ZAMBRETTI_I[]       = "Tidlig lett nedbør, senere oppklarnende vær";
  const char TEXT_ZAMBRETTI_J[]       = "Urolig, mulighet for lettere vær";
  const char TEXT_ZAMBRETTI_K[]       = "Stort sett oppholdsvær, sannsynligvis lette byger";
  const char TEXT_ZAMBRETTI_L[]       = "Ganske urolig, oppklarnende vær";
  const char TEXT_ZAMBRETTI_M[]       = "Urolig, mulighet for lettere vær";
  const char TEXT_ZAMBRETTI_N[]       = "Lett nedbør, noe sol";
  const char TEXT_ZAMBRETTI_O[]       = "Lett nedbør, senere urolig";
  const char TEXT_ZAMBRETTI_P[]       = "Vekslende vær, noe nedbør";
  const char TEXT_ZAMBRETTI_Q[]       = "Urolig, noe oppholdsvær";
  const char TEXT_ZAMBRETTI_R[]       = "Urolig, senere nedbør";
  const char TEXT_ZAMBRETTI_S[]       = "Urolig, noen byger";
  const char TEXT_ZAMBRETTI_T[]       = "Veldig urolig, noe oppholdsvær";
  const char TEXT_ZAMBRETTI_U[]       = "Noen byger, senere mer nedbør";
  const char TEXT_ZAMBRETTI_V[]       = "Noen byger, blir ganske urolig";
  const char TEXT_ZAMBRETTI_W[]       = "Byger";
  const char TEXT_ZAMBRETTI_X[]       = "Veldig urolig, nedbør";
  const char TEXT_ZAMBRETTI_Y[]       = "Uvær, mulighet for lettere vær";
  const char TEXT_ZAMBRETTI_Z[]       = "Uvær, store nedbørsmengder";
  const char TEXT_ZAMBRETTI_0[]       = "Tomt batteri, vennligst lad!";
  const char TEXT_ZAMBRETTI_DEFAULT[] = "Beklager, ikke noe værvarsel for øyeblikket";
#endif

3KUdelta, 2023-11-05:
Hello Markus, just try to delete the full line with translate();
Marc

3KUdelta, 2023-11-05:
Hello Markus, sorry I messed up V2.3 and V2.4 - they have different translation tables. In your case, just use V2.3 out of the history directory [link](https://github.com/3KUdelta/Solar_WiFi_Weather_Station/blob/master/history/Solar-WiFi-Weather-Station-V2.3.ino) This should work. I will clean up soon. 
Marc

3KUdelta, 2023-11-15:
Problem solved?

closed 2023-11-24 by 3KUdelta