3KUdelta/Solar_WiFi_Weather_Station #56: ThingSpeak and 18d20 temperature sensor

polanskiman, 2024-07-28:
Hello,

This is not an issue per say, more of a request.

I saw in the latest update v2.41 there is new code to improve the lifespan of the flash memory. I would like to take advante of that and also of all the updates you have made through the years (my Station is currently running with v2.35). However I see that all ThingSpeak code was removed since v2.4. I have been trying to use old code to add it to v2.41 but I am failing misarably. Any chance that it can be added back?

Also I noticed that you are using the 18d20 temperature sensor but I am using the BME280 sensor. Do I need to comment out some code?

Forgive my ignorance, I'm not really a coder.

3KUdelta, 2024-10-21:
Dear polanskiman,
thank you for your valuable comment. I am sorry about dropping the ThingSpeak part. It should really be a copy/paste exercise since there were no changes in the meantime on ThingSpeak's side. 
I recommend to buy a 18d20. The ones which are coming in a sealed metal tube (something like this: https://de.aliexpress.com/item/32342208694.html). It is much less sensitive to direct sunlight and gives the system some inertia. We even added a hdc1080 which can be heated in order to get better humidity readings. The BME280 is good for pressure but unfortunately very unreliable for humidity. It gets stuck to 100% quite often. The 10d20 plugs into D7 and needs a 4.7k resistor soldered to positive and data to pull it high. The hdc1080 into the I2C socket (the one with SCL and SCA).
Marc

closed 2025-03-21 by 3KUdelta

phit, 2025-03-21:
hey @polanskiman, I added back Thingspeak support as I was also using it and there's now configs to disable 18d20 so you should be able to update now 
otherwise you can also apply the fix to 2.3X, the only really important change is the removal of this useless write on no change

https://github.com/3KUdelta/Solar_WiFi_Weather_Station/commit/16c5c59779aadb1d691e766517d1a071fb625812#diff-b0db574389ed001b0001dba802a444cb1cbcfa4e2337835968cbc2e16c0ebad7L304-L306

PR #57 (2025-03-21): Readd Thingspeak; restructure settings
- readd Thingspeak
- move minimum battery voltage & calibration to settings
- remove unused uploaders during compile
- remove unused blynk settings

closes #56